openwrt/luci #1711: jsonrpc broken in trunk

sophana, 2018-04-01:
json rpc works in 17.01, no more in trunk
auth works ok, but any request returns code 403.
ubus call session list shows normal session.

tested in snapshot, then installed luci-base luci-mod-rpc

I'd like to help debug this, but when I try do make prints inside the rpc.lua, I don't see anything.

sophana, 2018-04-01:
I found the problem in dispatcher.lua.
Below fix that is working.
```$ git diff
diff --git a/modules/luci-base/luasrc/dispatcher.lua b/modules/luci-base/luasrc/dispatcher.lua
index 16b3254..e369125 100644
--- a/modules/luci-base/luasrc/dispatcher.lua
+++ b/modules/luci-base/luasrc/dispatcher.lua
@@ -175,7 +175,7 @@ local function session_retrieve(sid, allowed_users)
           type(sdat.values) == "table" and
           type(sdat.values.token) == "string" and
           (not allowed_users or
-           util.contains(allowed_users, sdat.values.username))
+           util.contains(allowed_users, sdat.values.user))
        then
                return sid, sdat.values
        end
@@ -426,7 +426,7 @@ function dispatch(request)
 
                ctx.authsession = sid
                ctx.authtoken = sdat.token
-               ctx.authuser = sdat.username
+               ctx.authuser = sdat.user
        end
 
        if c and require_post_security(c.target) then
```

sophana, 2018-04-01:
finally made the change into rpc and adopted username field instead of user in rpc.
Made a pull request #1713

jow-, 2018-04-06:
Fixed with feefc60

closed 2018-04-06 by jow-