openwrt/luci #1720: Reflected XSS in 'redir'-parameter

macintorsten, 2018-04-04:
First, login, if needed

1. Go to http://ip-or-hostname/cgi-bin/luci/admin/uci/changes?redir=javascript:alert(document.cookie)
2. Click on 'Back'

**Result**: Pop up the the message 'sysauth=xxxxxxxxxxxxxxxxxxxxxxxxxx' i.e XSS POC.

This issue is less severe than a normal reflected XSS because of the following reasons:

- User action is required to trigger the XSS
- `X-Frame-Options: SAMEORIGIN` should prevent click-jacking attacks. Fixed see #1343 

Allowing relative URLs in this kind of parameter is very error prone. Also could the `secure` attribute be added to the sysauth cookie so an XSS can't leak the cookie value? See https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Set-Cookie

Tested on:
> Powered by LuCI Master (git-18.088.49307-90ed423) / OpenWrt SNAPSHOT r6567-db893ec

macintorsten, 2018-04-05:
Here is a very similar one, but in step 2, click on "Reset" button:

http://192.168.56.2/cgi-bin/luci/admin/system/packages?exec=0&display=%27-alert(1)//&query=_

jow-, 2018-04-05:
Case 1: https://github.com/openwrt/luci/commit/7b04d0bbcf0f34393f20ccad8884a67fea9e2863, https://github.com/openwrt/luci/commit/731ed77c0bbee7004a6b5645d9a8592a76748a1c
Case 2: https://github.com/openwrt/luci/commit/bfc98bec4d79efb24434e11ea27b3c17c31365ab

closed 2018-04-14 by macintorsten