openwrt/luci #1722: Security updates: Simpleform does no longer works?

dibdot, 2018-04-05:
@jow- Maybe you can have a look. With the latest security updates in place I cannot use SimpleForm any longer, e.g. blacklist_tab.lua in adblock package:

[image]

Thanks!

jow-, 2018-04-05:
This seems to be caused by some SimpleForm template markup issues. In my local test I see that the beginning of the form is rendered twice, like this:
```

<form method="post" name="cbi" action="/cgi-bin/luci/admin/services/adblock/advanced" enctype="multipart/form-data" onreset="return cbi_validate_reset(this)" onsubmit="return cbi_validate_form(this, 'Some fields are invalid, cannot save values!')">
	<div>
		<script type="text/javascript" src="/luci-static/resources/cbi.js?v=git-18.071.69548-de88c2e" data-strings="{&#34;path&#34;:{&#34;resource&#34;:&#34;\/luci-static\/resources&#34;,&#34;browser&#34;:&#34;\/cgi-bin\/luci\/admin\/filebrowser&#34;},&#34;label&#34;:{&#34;choose&#34;:&#34;-- Please choose --&#34;,&#34;custom&#34;:&#34;-- custom --&#34;}}"></script>
		<input type="hidden" name="token" value="63c1ed2e026a54ffbaa2365498422146" />
		<input type="hidden" name="cbi.submit" value="1" />
		<input type="submit" value="Save" class="hidden" />
	</div>

<form method="post" enctype="multipart/form-data" action="/cgi-bin/luci/admin/services/adblock/advanced">
	<div>
		<script type="text/javascript" src="/luci-static/resources/cbi.js?v=git-18.071.69548-de88c2e"></script>
		<input type="hidden" name="token" value="63c1ed2e026a54ffbaa2365498422146" />
		<input type="hidden" name="cbi.submit" value="1" />
	</div>

	<div class="cbi-map" id="cbi-input">
```

This causes the token value to be sent twice, causing the verification to fail - will look into it asap.

jow-, 2018-04-05:
I took the liberty to commit a fix for this directly. The redundant markup rendering was caused by instantiating the SimpleForm models through the `cbi()` and not the `form()` target.

wolandmaster, 2021-03-05:
I don't know whether it is the intended way to use it, but this is working:
```_G.redirect = luci.dispatcher.build_url("...")```