openwrt/luci #1725: Latest master not working

mcaptur, 2018-04-06:
I get this when i access router with latest master

/usr/lib/lua/luci/controller/admin/uci.lua:8: attempt to index global 'disp' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/admin/uci.lua:8: in function 'v'
	/usr/lib/lua/luci/dispatcher.lua:598: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:256: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

mcaptur, 2018-04-06:
reverting 
@@ -5,8 +5,7 @@
 module("luci.controller.admin.uci", package.seeall)
 
 function index()
-	local redir = luci.http.formvalue("redir", true) or
-	  luci.dispatcher.build_url(unpack(luci.dispatcher.context.request))
+	local redir = luci.http.formvalue("redir", true) or table.concat(disp.context.request, "/")
 
 	entry({"admin", "uci"}, nil, _("Configuration"))
 	entry({"admin", "uci", "changes"}, call("action_changes"), _("Changes"), 40).query = {redir=redir}

fixes it

luaraneda, 2018-04-06:
Confirmed, both the issue and the fix from @mcaptur 

Run tested on an ASUS RT-AC58U (ipq40xx), OpenWRT^c089671, luci^bfc98be

closed 2018-04-06 by jow-

mcaptur, 2018-04-06:
Now when trying to wake a host from luci WOL addon i get the following 

Form token mismatch

The submitted security token is invalid or already expired!

In order to prevent unauthorized access to the system, your request has been blocked. Click "Continue »" below to return to the previous page.

not sure if related

jow-, 2018-04-06:
No, it is a different issue, fixed with dacf88216d93919d8e988a08902c9f87779874ee

dibdot, 2018-04-06:
@jow- What's the recommended way to bring back the "cbi-magic" after the switch to the form() action? E.g. in the wol app something like this no longer works ... same applies to field dependencies etc.

```
host = s:option(Value, "mac", translate("Host to wake up"),
        translate("Choose the host to wake up or enter a custom MAC address to use"))

sys.net.mac_hints(function(mac, name)
        host:value(mac, "%s (%s)" %{ mac, name })
end)
```

commit d6b4a3d4 (2018-04-06): luci-mod-admin-full: fix request path access in uci controller

Fixes #1725
Fixes 731ed77c0 ("treewide: improve handling of page redirections in uci change views")