openwrt/luci #1726: mwan error

fuqiang03, 2018-04-06:
[image]

feckert, 2018-04-06:
@fuqiang03 what version do you use openwrt-15.05 lede-17.01 or master?

jow-, 2018-04-06:
@feckert - I'm on it

closed 2018-04-06 by jow-

feckert, 2018-04-06:
@jow- I have seen your change in the commit history 24ac0d81ee56c5630b38aefbb5526e8bfcf6cd88
Please let me know what i have to change in the sources?

jow-, 2018-04-06:
The main problem is twofold:

1) mwan3 relies on deprecated uci network state variables instead of using the ubus runtime state, this can be easily fixed (will provide an example change in a later comment)
2) mwan3 saves its own runtime state in the form of uci state variables

The problem with uci state variables is that they cannot be clearly differentiated from config values, means they will shadow values from /etc/config/ but it is not possible to figure out if an effective value is either a state value or a configured one.

There also is no defined API way to only query state variables without merging /etc/config/ information, and finally the /var/state directory is globally shared which prevents us from implementing proper uci session isolation in future LuCI commits.

Ideally we need to do two things:
1) change the mwan3 network state bits to use ubus info, this can be easily done using the LuCI network model
2) refactor mwan3 itself to either not use /var/state uci info or extend it with some kind of `mwan3 get_state ...` command which can be queried from LuCI

jow-, 2018-04-06:
@feckert - here's an exemplary change to rewrite the mwan3 interface tab using the uci and network models. Only uci state access remaining is the one to mwan3's own private variables.

```diff
diff --git a/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua b/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua
index 8564231c8..3f1389276 100644
--- a/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua
+++ b/applications/luci-app-mwan3/luasrc/model/cbi/mwan/interface.lua
@@ -2,8 +2,9 @@
 -- Copyright 2018 Florian Eckert <fe@dev.tdt.de>
 -- Licensed to the public under the GNU General Public License v2.
 
-dsp = require "luci.dispatcher"
-uci = require "uci"
+local dsp = require "luci.dispatcher"
+local uci = require "luci.model.uci".cursor()
+local ntm = require "luci.model.network".init()
 
 
 function interfaceWarnings(overview, count, iface_max)
@@ -49,59 +50,36 @@ function interfaceWarnings(overview, count, iface_max)
 end
 
 function configCheck()
-	local overview = {}
 	local count = 0
+	local overview = {}
 	local duplicate_metric = {}
-	uci.cursor():foreach("mwan3", "interface",
+
+	uci:foreach("mwan3", "interface",
 		function (section)
-			local uci = uci.cursor(nil, "/var/state")
 			local iface = section[".name"]
-			overview[iface] = {}
-			count = count + 1
-			local network = uci:get("network", iface)
-			overview[iface]["network"] = false
-			if network ~= nil then
-				overview[iface]["network"] = true
-
-				local device = uci:get("network", iface, "ifname")
-				if device ~= nil then
-					overview[iface]["device"] = device
-				end
-
-				local metric = uci:get("network", iface, "metric")
-				if metric ~= nil then
-					overview[iface]["metric"] = metric
-					overview[iface]["duplicate_metric"] = false
-					for _, m in ipairs(duplicate_metric) do
-						if m == metric then
-							overview[iface]["duplicate_metric"] = true
-						end
-					end
-					table.insert(duplicate_metric, metric)
-				end
-
-				local dump = require("luci.util").ubus("network.interface.%s" % iface, "status", {})
-				overview[iface]["default_route"] = false
-				if dump and dump.route then
-					local _, route
-					for _, route in ipairs(dump.route) do
-						if dump.route[_].target == "0.0.0.0" then
-							overview[iface]["default_route"] = true
-						end
-					end
+			local state = { network = false }
+			local network = ntm:get_network(iface)
+			local trackingNumber = uci:get_state("mwan3", iface, "track_ip")
+			local reliabilityNumber = tonumber(uci:get_state("mwan3", iface, "reliability"))
+
+			if network then
+				state.network = true
+				state.device = network:ifname()
+				state.default_route = (network:gwaddr() ~= nil)
+
+				local metric = network:metric()
+				if metric > 0 then
+					state.metric = metric
+					state.duplicate_metric = duplicate_metric[metric] or false
+					duplicate_metric[metric] = true
 				end
 			end
 
-			local trackingNumber = uci:get("mwan3", iface, "track_ip")
-			overview[iface]["tracking"] = 0
-			if trackingNumber and #trackingNumber > 0 then
-				overview[iface]["tracking"] = #trackingNumber
-				overview[iface]["reliability"] = false
-				local reliabilityNumber = tonumber(uci:get("mwan3", iface, "reliability"))
-				if reliabilityNumber and reliabilityNumber <= #trackingNumber then
-					overview[iface]["reliability"] = true
-				end
-			end
+			state.tracking = trackingNumber and #trackingNumber or 0
+			state.reliability = (reliabilityNumber and (reliabilityNumber < #trackingNumber) or false)
+
+			overview[iface] = state
+			count = count + 1
 		end
 	)
 
@@ -116,13 +94,11 @@ function configCheck()
 		return hasbit(x, p) and x or x + p
 	end
 
-	local uci = require("uci").cursor(nil, "/var/state")
-	local mmx_mask = uci:get("mwan3", "globals", "mmx_mask") or "0x3F00"
-	local number = tonumber(mmx_mask, 16)
 	local bits = 0
 	local iface_max = 0
+	local mmx_mask = tonumber(uci:get_state("mwan3", "globals", "mmx_mask") or "", 16) or 0x3F00
 	for i=1,16 do
-		if hasbit(number, bit(i)) then
+		if hasbit(mmx_mask, bit(i)) then
 			bits = bits + 1
 			iface_max = setbit( iface_max, bit(bits))
 		end
@@ -228,10 +204,10 @@ end
 metric = mwan_interface:option(DummyValue, "metric", translate("Metric"))
 metric.rawhtml = true
 function metric.cfgvalue(self, s)
-	local uci = uci.cursor(nil, "/var/state")
-	local metric = uci:get("network", s, "metric")
-	if metric then
-		return metric
+	local network = ntm:get_network(s)
+	local metric = network and network:metric() or 0
+	if metric > 0 then
+		return tostring(metric)
 	else
 		return "&#8212;"
 	end
```

Ideally I want to get rid of the few remaining uci:get_state() calls entirely, maybe by replacing them with some `luci.util.exec("mwan3 get_state whatever")` or similar.

feckert, 2018-04-06:
@jow- Thanks for clarify this.
I am not sure if i have time to fix this for the upcoming owrt release 18.05!
Because to less testing time and i am on the year vacation next 3 weeks.
If i do **not** change this will the actual source work in the next release?
Are your changes ready to apply if so could you change this?
I will come back to fix this after the release in the new master -> if you have not enough time.

jow-, 2018-04-06:
@feckert - after the fixes currently in master it will work for the forseeable future. It is just LuCI itself which stopped using `libuci-lua` hence the missing global `uci` leading to the error originally reported here.

Applications that still require direct uci features need to import `libuci-lua` (`require "uci"`) themselves in order to use it like before.

Refactoring mwan3 can wait until after the release.

fuqiang03, 2018-04-06:
Need to be updated?

commit 08a2b27d (2018-04-06): luci-app-mwan3: fix legacy uci api usage

Explicitely require libuci-lua in model classes that use legacy /var/state
cursor handling.

Also add a specific dependency on libuci-lua to the luci-app-mwan3
Makefile in preparation of the upcoming default removal of libuci-lua.

Finally fix the post data dispatching on the notification tab, see #1722
for reference.

Fixes #1726.