openwrt/luci #1732: Dispatcher errors for entries in '/admin/system/advanced_reboot'

VindicatorDS, 2018-04-07:
luci-app-advanced-reboot returns the following error when performing a shutdown through the GUI:

/usr/lib/lua/luci/dispatcher.lua:495: Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/power_off'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:197: attempt to index global 'uci' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:495: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

luci-app-advanced-reboot also returns the following error when performing a partition switch through the GUI:

/usr/lib/lua/luci/dispatcher.lua:495: Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/alternative_reboot'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:165: attempt to index global 'uci' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:495: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

I'm using buildbot images and packages. My latest build was using luci git-18.088.49307-90ed423 and was OK. I'm now getting this error with luci git-18.096.59853-52ceba1-1

hnyman, 2018-04-07:
Probably related to recent changes and advanced reboot needs similar fix as some other packages, e.g. discussed here https://github.com/openwrt/luci/issues/1726#issuecomment-379254050

This fixes both reboots for me:
```
--- /rom//usr/lib/lua/luci/controller/advanced_reboot.lua
+++ /usr/lib/lua/luci/controller/advanced_reboot.lua
@@ -3,6 +3,8 @@

 module("luci.controller.advanced_reboot", package.seeall)

+uci = require "uci"
+
 -- device_name, board_name, part1, part2, offset, env_var_1, value_1_1, value_1_2, env_var_2, value_2_1, value_2_2
 devices = {
   {"Linksys EA3500", "linksys-audi", "mtd3", "mtd5", 32, "boot_part", 1, 2, "bootcmd", "run nandboot", "run altnandboot"},

```

But @jow- knows better than me if that is the exactly correct fix.
cc also @stangri

stangri, 2018-04-07:
Thanks Hannu.

I've fixed it in my repo (packages available at https://stangri.github.io/openwrt-repo/ and the source is available at https://github.com/stangri/openwrt_packages), I'll send PRs for affected packages no later than Monday.

hnyman, 2018-04-07:
If you have them ready, please send them sooner.

(based on mailing discussion, 18.05 branching might be as soon as today. Not sure if that will happen, but if would be great to avoid the need for backporting)

hnyman, 2018-04-07:
reference to timeline
https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg41769.html

and you can send them as one PR, as it is about a known common issue

VindicatorDS, 2018-04-08:
1104b837cdf87e42512a32c48c9bedbd9c1d6d57 (https://github.com/stangri/openwrt_packages/commit/c7c46a9afae8fa1360be7c80ac8c3c3bbb121c3c) fixed it. Thanks!

closed 2018-04-08 by VindicatorDS