openwrt/luci #1736: software package

knaters, 2018-04-07:
Any help with this?
Trying to install package, using LuCI Master (git-18.096.73721-9304042)
Need this to get info of each IP connected, data use etc.

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-nlbwmon:
 * 	kernel (= 4.9.91-1-32f977e15ddea593ee4fcb1dca2e5862) * 	kernel (= 4.9.91-1-32f977e15ddea593ee4fcb1dca2e5862) * 
 * opkg_install_cmd: Cannot install package luci-app-nlbwmon.

dibdot, 2018-04-07:
Please see here: https://openwrt.org/faq/after_installation#cannot_satisfy_dependencies

In practise if you need custom kernel modules for your snapshot, you need to compile the whole kernel/firmware at the same time. Or use the latest release ...

closed 2018-04-07 by dibdot

knaters, 2018-04-07:
fixed the problem.
updated my snapshot and installed latest luci .
everything ok now