openwrt/luci #1745: omap platform Flash new firmware image is fail  and not support

guhongyeying, 2018-04-11:
1：i can find  “openwrt/target/linux/x86/base-files/lib/upgrade/platform.sh” and  platform Flash new firmware image is success
2：but  linux omap not find  “openwrt/target/linux/x86/base-files/lib/upgrade/platform.sh” and error"Sorry, there is no sysupgrade support present; a new firmware image must be flashed manually. Please refer to the OpenWrt wiki for device specific install instructions."
3： who can tell me How to do it。

dibdot, 2018-04-11:
This has nothing to do with LuCI GUI - please check the wiki and/or ask in the forum (https://forum.lede-project.org) for support.

closed 2018-04-11 by dibdot

guhongyeying, 2018-04-11:
1：i can find   linux x86 “openwrt/target/linux/x86/base-files/lib/upgrade/platform.sh” and platform Flash new firmware image is success
2：but linux omap not find “openwrt/target/linux/omap/base-files/lib/upgrade/platform.sh” and error"Sorry, there is no sysupgrade support present; a new firmware image must be flashed manually. Please refer to the OpenWrt wiki for device specific install instructions."
3:why?

guhongyeying, 2018-04-11:
1 : linux x86   platform Flash new firmware image is success
2 : but linux omap platform Flash new firmware image is fail

guhongyeying, 2018-04-11:
thanks you @dibdot