openwrt/luci #1762: lxc container not working

Eddiecz, 2018-04-23:
flashed 
turris-omnia-sysupgrade.img.gz | 21f37514940697f9d88f9069748c555e0d217c2464cfc0998f7d79f95e57cb26 | 5101.1 KB | Mon Apr 23 17:26:02 2018

Firmware Version | OpenWrt SNAPSHOT r6745-b1f5dd3 / 			LuCI Master (git-18.112.36203-82abf9a)

Installed luci and luci-app-lxc

Containers URL is images.linuxcontainers.org , but no containers shown. Changed to uk.images.linuxcontainers.org , same result.

localnet, 2018-04-28:
Are you waiting enough time to download the image list? Can you try to execute the next command from the console?

```
sh /usr/share/lxc/templates/lxc-download --list --no-validate --server images.linuxcontainers.org
```

Eddiecz, 2018-04-28:
I tried command `sh /usr/share/lxc/templates/lxc-download --list --no-validate --server images.linuxcontainers.org`  that was okay.
The last state was that the containers appeared, but is impossible create container in luci.
I've tried lxc-create command, it ended in error, so it's probably a problem in lxc itself.
Now i can not test, turris is  back on original firmware.
Openwrt has DSA, original swconfig, but I need Vlan and tagging.

Eddiecz, 2018-05-10:
any progressa about this problem? I am testing x86 image in virtualbox, and problem persist, no containers shown in luci :-(

PR #1808 (2018-05-23): luci-app-lxc: fix "plain-vanilla" integration
I've tried to get the lxc app in a more usable state. Tested with mips
and amd64 targets.

* fix missing tar/xz dependency,
  only if LXC_BUSYBOX_OPTIONS is not selected
* mute needless gpg validation warning
* tidy up controller a little bit
* fix multiple possible dispatcher errors
* fix compatibility with XHTML standard theme (looks still horrible ;-)
* inform the user about custom kernel prerequisites
* inform the user about the template download
* inform the user if no template was found

Should fix #1762 and other complains in forum / package [... cut]
closed 2018-05-23 by dibdot