openwrt/luci #1770: Incorrect section used on Save & Apply from Wireless Interface Configuration

kristrev, 2018-04-26:
Hello,

I have been debugging an issue where connecting to a wifi network using Luci is failing. My steps for connecting/configuring are as follows:

* Go to the Wireless page of Luci.
* Click Scan, Join Network and then configure network.
* Click Submit and then Save & Apply on the next page.

After clicking Save & Apply, my wifi (wifi-iface) configuration is wiped and wifi is left in a non-working state.

I did some digging, and it seems the problem is in line 27 of luci-mod-admin-full/luasrc/model/cbi/admin_network/wifi.lua. When adding some debug output, I see that wnet.sid contains the section name of the section that is to be deleted. If I manually change ifsection.section to the new section name (between "Submit" and "Save & Apply"), wifi is configured properly. I also noticed that ifsection already contains a section-variable when we enter this conditional, and the value of section is that of the new config.

Bisecting Luci led me to the following commit: 4024d4f2242a88a5e9fd18ed437b1c9fd13693f9. Before this commit, configuring wifi works fine. However, after this commit I can see the error.

jow-, 2018-04-26:
I assume you used the `Replace configuration` checkbox?

jow-, 2018-04-26:
Can you provide a screenshot of the `Unsaved changes` page (open in a new tab) before hitting `Save & Apply` and can you attach the generated HTML markup of the wireless settings page?

kristrev, 2018-04-26:
Thanks a lot for the quick response!

Yes, I use the replace configuration checkbox, should have mentioned that. I have attached the screenshot + html. I had to change the ending of the html to txt to overcome some Github restriction :)

[image]
[wireless-settings.txt](https://github.com/openwrt/luci/files/1951864/wireless-settings.txt)

jow-, 2018-04-27:
This matches what I see here but so far I was unable to reproduce the problem with LuCI master. Can you also provide me with trashed /etc/config/wireless?

jow-, 2018-05-06:
The bad news is that I have no idea yet whats going on, the good news is that I was finally able to reproduce it on a tl-wr703n :)

closed 2018-05-19 by jow-

kristrev, 2018-06-14:
Hi,

I was doing some testing and I still see this error with the latest master :( The steps for reproducing is the same.

commit fdbb417a (2018-05-19): luci-mod-admin-full: do not override iface section in wireless cbi map

Do override the iface section id upon commit to avoid clobbering the resulting
configuration. The manual config sync is not needed anymore after switching
to uncached ubus uci operations.

Fixes #1770.