openwrt/luci #1772: Lower Case MAC Addr in Static Leases (luci/admin/network/dhcp)

VindicatorDS, 2018-04-29:
Hosts present in /etc/config/dhcp with upper case mac addresses and option name are correctly shown by Luci in luci/admin/network/dhcp under static leases, with the "MAC-Address" column with the format "MAC-Addr (hostname.localdomain)", e.g., "00:26:BF:37:50:76 (switch.mydomain)".

For hosts configured with lower case mac addresses (or a mix of lower and upper case letters), Luci only shows the mac-addr under static leases (MAC-Address column), this is, the (hostname.localdomain) is not shown, e.g., "00:26:bf:37:50:76".

upper-case MAC
[image]

lower-case MAC
[image]

I'm using luci git-18.118.21533-00d2429. I don't know if this is a new or an old issue (or a no issue if by design...).

jow-, 2018-05-02:
There is no commit `00d2429` in the LuCI repository. Where did you obtain your packages from?

closed 2018-05-02 by jow-

VindicatorDS, 2018-05-02:
Tnx for the fix.

luci git-18.118.21533-00d2429 (https://github.com/openwrt/luci/commit/00d242918f7de007f2a05f102a9660074013293f) was obtained from buildbot build #1188 (https://phase2.builds.lede-project.org/builders/arm_cortex-a9_vfpv3/builds/1188). It accompanied OpenWRT r6775.

commit 9adbd79f (2018-05-02): luci-mod-admin-full: canonicalize configured static lease MAC

When reading the configured mac address of the static lease, filter it
through luci.ip.checkmac() to canonicalize and uppercase the value for
mapping it against the combo box host hints.

Fixes #1772.