openwrt/luci #1779: network.lua:1277: bad argument #4 to '?'

john-shine, 2018-05-05:
When try to visit cgi-bin/luci/admin/network/wireless_join, i get error:
lan: 500 Internal Server Error
Sorry, the server encountered an unexpected error.

/usr/lib/lua/luci/dispatcher.lua:507: Failed to execute call dispatcher target for entry '/admin/network/wireless_join'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/simpleform'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nullsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/firewall_zonelist'.
A runtime error occured: /usr/lib/lua/luci/model/network.lua:1277: bad argument #4 to '?' (no value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:507: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function 

I try to debug, but edit code has no effect

jow-, 2018-05-05:
Please provide the output of `ubus call network.wireless dump` and `uci show wireless` as well `uci show network`.

jow-, 2018-05-19:
I added a workaround which should cure the crash. Reopen if you still see this.

closed 2018-05-19 by jow-

john-shine, 2018-05-23:
Thanks, i will try this later.