openwrt/luci #1781: "Override MTU" for a Guest WiFi Interface

fantom-x, 2018-05-06:
Setting "Override MTU" for a guest WiFi network interface does not take effect after **Save & Apply**; a reboot is required to see a new MTU on the interface. The same change for a wired interface or a bridge does not require a reboot and is active right after **Save & Apply**.

hnyman, 2018-05-06:
Not quite sure if this is LuCI issue, or an issue of the underlying wifi driver components (driver, hostapd, etc.).

with exactly which LuCI version?
17.01? master?  (or master with new rollback code?)

fantom-x, 2018-05-06:
The latest master built earlier today, but I not think that is a new issue though; I saw it a while back.

fantom-x, 2018-05-06:
The value is saved, it just does not take effect until a reboot. Maybe something does not get restarted?

hnyman, 2018-05-06:
> The latest master built earlier today

That is the reason why I was asking about the exact version. 
jow has been rewriting some of the code, so it is possible that this is rather recent regression. 

Most of the wifi configuration will get revamped along the rollback functionality. see https://github.com/openwrt/luci/pull/1769/commits for the current patches.

Similar symptoms were discussed e.g. started from https://github.com/openwrt/luci/pull/1769#issuecomment-385228343 but that is mostly in the context of the first version of rollback. But still, the old wifi section code has some peculiarities, so a rewrite of that code may be right on the spot.

jow-, 2018-07-03:
That needs to be fixed in OpenWrt. LuCI relies on netifd/procd doing the right things on reload.

closed 2018-07-03 by jow-