openwrt/luci #1784: Get the response "Bad Gateway" while accessing web server

garywangcn, 2018-05-08:
I compiled Openwrt with the latest luci code, the configuration for LUCI module is below:

> CONFIG_PACKAGE_liblucihttp=y
> CONFIG_PACKAGE_liblucihttp-lua=y
> CONFIG_PACKAGE_luci=y
> CONFIG_PACKAGE_luci-ssl-openssl=y
> CONFIG_PACKAGE_luci-base=y
> CONFIG_PACKAGE_luci-mod-admin-full=y
> CONFIG_PACKAGE_luci-app-ddns=y
> CONFIG_PACKAGE_luci-app-firewall=y
> CONFIG_PACKAGE_luci-app-lxc=y
> CONFIG_PACKAGE_luci-app-qos=y
> CONFIG_PACKAGE_luci-theme-bootstrap=y
> CONFIG_PACKAGE_luci-theme-openwrt=y
> CONFIG_PACKAGE_luci-proto-ipv6=y
> CONFIG_PACKAGE_luci-proto-ppp=y
> CONFIG_PACKAGE_luci-lib-httpclient=y
> CONFIG_PACKAGE_luci-lib-httpprotoutils=y
> CONFIG_PACKAGE_luci-lib-ip=y
> CONFIG_PACKAGE_luci-lib-jsonc=y
> CONFIG_PACKAGE_luci-lib-nixio=y
> 

But when I program image to target and access it via chrome(https://192.168.1.1/cgi-bin/luci), get the follow information:
    Bad Gateway
    The process did not produce any response

Do you someone encounter this issue? Is there a way to see what problem happens?

cdbierl, 2018-05-09:
Yeah I have the same behavior. I see nothing in the logs to explain why though.

garywangcn, 2018-05-15:
Please see this link https://github.com/jow-/lucihttp/issues/1  to get more information.

commit 27b7e431 (2018-05-18): lucihttp: update to latest HEAD

6ddea4c utils: fix crash with zero length input string in lh_header_attribute()
cb119de lib: add support for setting the maximum allowed data size

Fixes #1784.
closed 2018-05-18 by jow-