openwrt/luci #1795: Mount Points / Mounted file systems: Styling issue

he-ma, 2018-05-15:
There seems to be a little style problem which takes effect on certain themes. The attached picture should show what I mean. If no button is rendered (for system mount points) the table is truncated at the last field. Would be more nicer if all table rows have the same length.

[image]

Code position where the button is created:
https://github.com/openwrt/luci/blob/f3b30512361584c61c9daa51650ad833072aa919/modules/luci-mod-admin-full/luasrc/model/cbi/admin_system/fstab.lua#L119-L125

The view which is used for the button:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/view/cbi/button.htm

jow-, 2018-05-15:
This is a bug with the material theme I think. Iirc it has some global CSS rule to hide all empty fields, which would apply to empty `td` as well.

he-ma, 2018-06-05:
Hello @jow-
I did a research again and found out that the real problem is that there is no \<td\>\</td\> at the end of the line nor is a "hidden" button created if no button rendering is done (in button.htm).

So I modified button.htm to the following:

```
<%+cbi/valueheader%>
	<% if self:cfgvalue(section) ~= false then %>
		<% if self.inputstyle ~= "hidden" then %>
			<input class="cbi-button cbi-input-<%=self.inputstyle or "button" %>" type="submit"<%= attr("name", cbid) .. attr("id", cbid) .. attr("value", self.inputtitle or self.title)%> />
		<% else %>
			<input class="cbi-button cbi-input-button" type="hidden" />
		<% end %>
	<% else %>
		-
	<% end %>
<%+cbi/valuefooter%>

```
And modifying fstab.lua (here starting at line 119) to the following:

```
unmount = v:option(Button, "unmount", translate("Unmount"))
unmount.render = function(self, section, scope)
	if non_system_mounts[section].umount then
		self.title = translate("Unmount")
		self.inputstyle = "remove"
        	Button.render(self, section, scope)
	else
		self.inputstyle = "hidden"
		Button.render(self, section, scope)
	end
end

```
This works.

Maybe it's possible to extend the button class to use a type argument which defaults to submit, but can be modified to hidden? Alternatively could you kindly merge these modifications into the source?

feckert, 2019-09-16:
Fixed by https://github.com/openwrt/luci/commit/d79d9874d7a1cf29531ee927b2668263557b21f3

closed 2019-09-16 by feckert