openwrt/luci #1802: Hand in overview page

Ansuel, 2018-05-18:
I have this error... I can only see system information from RAM section all is loading in progress...

here the error

overview:127 Uncaught TypeError: Cannot read property 'wan' of null
    at overview:127
    at XMLHttpRequest.xhr.onreadystatechange (xhr.js?v=git-18.138.35484-521554f:75)

@jow-

jow-, 2018-05-19:
Please provide the output of `http://192.168.1.1/cgi-bin/luci?status=1` (change IP address accordingly)

Ansuel, 2018-05-19:
{"swap":{"free":33550336,"total":33550336},"conncount":725,"leases":[{"expires":32717,"macaddr":"30:07:4D:1B:EE:6C","ipaddr":"192.168.2.101","hostname":"Galaxy-S8"},{"expires":27602,"macaddr":"9C:B6:D0:EC:F5:45","ipaddr":"192.168.2.100","hostname":"Ansuel-XPS"}],"leases6":[],"memory":{"buffered":9936896,"total":490807296,"shared":2965504,"free":374284288},"uptime":10622,"wifinets":[{"device":"radio0","networks":[{"ifname":"wlan0","disabled":false,"encryption":"mixed WPA\/WPA2 PSK (CCMP)","bssid":"DC:EF:09:F2:5D:51","mode":"Master","quality":77,"noise":-103,"ssid":"No More Lag 5Ghz","link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio0.network1","assoclist":{"9C:B6:D0:EC:F5:45":{"rx_short_gi":true,"noise":-103,"rx_mcs":9,"rx_vht":true,"rx_mhz":80,"rx_nss":2,"tx_vht":false,"tx_ht":false,"tx_rate":22400,"tx_packets":813444,"tx_mhz":20,"rx_packets":473649,"inactive":2,"rx_ht":false,"rx_rate":866700,"signal":-63},"30:07:4D:1B:EE:6C":{"tx_mcs":9,"noise":-103,"tx_ht":false,"rx_vht":false,"rx_mhz":20,"tx_nss":2,"tx_short_gi":true,"tx_vht":true,"tx_rate":866700,"tx_packets":6731,"tx_mhz":80,"rx_packets":11037,"inactive":5074,"rx_ht":false,"rx_rate":24000,"signal":-49}},"txpoweroff":0,"bitrate":444.5,"txpower":30,"name":"Master \"No More Lag 5Ghz\"","channel":104,"country":"ZA","signal":-56,"up":true,"frequency":"5.520"}],"name":"Qualcomm Atheros QCA9984 802.11ac Wireless Controller (radio0)","up":false},{"device":"radio1","networks":[{"ifname":"wlan1","disabled":false,"encryption":"mixed WPA\/WPA2 PSK (CCMP)","bssid":"DC:EF:09:F2:5D:52","mode":"Master","quality":0,"noise":-90,"ssid":"No More Lag","link":"\/cgi-bin\/luci\/admin\/network\/wireless\/radio1.network1","assoclist":[],"txpoweroff":0,"txpower":30,"name":"Master \"No More Lag\"","channel":1,"country":"ZA","signal":0,"up":true,"frequency":"2.412"}],"name":"Qualcomm Atheros QCA9984 802.11bgn Wireless Controller (radio1)","up":true}],"wan":{"proto":"pppoe","ipaddr":"212.171.81.93","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan","netmask":"255.255.255.255","gwaddr":"192.168.100.1","expires":-1,"uptime":191,"ifname":"pppoe-wan","dns":["85.37.17.16","85.38.28.68"]},"localtime":"Sun May 20 01:40:41 2018","connmax":16384,"loadavg":[15808,17376,13344]}

Ansuel, 2018-05-19:
@jow-

jow-, 2018-05-20:
Is the output you see in the browser really preinted like this? `..."name":"Master "No More Lag 5Ghz""...`? Seems the escaping of the inner quotes is broken.

jow-, 2018-05-20:
Is the output of this command: `lua -lluci.jsonc -e 'print(luci.jsonc.stringify([[Master "No More Lag"]]))'` exactly `"Master \"No More Lag\""` ?

jow-, 2018-05-20:
Which browser do you use?

Ansuel, 2018-05-20:
Chrome

jow-, 2018-05-20:
Unfortunately I am unable to reproduce it here. Please check if you get any network errors or similar when open the debugger tools with F12.

Ansuel, 2018-05-21:
@jow- i think i found the problem...

time lua /www/cgi-bin/luci  output this time

real    0m 13.33s
user    0m 0.57s
sys     0m 0.18s

i'm on stock, no additional configuration flag... any idea to backtrack all this extra time and slowness?

closed 2018-05-23 by Ansuel