openwrt/luci #1803: LuCI 500 Internal Server Error: unable to find section 'main'

initramfs, 2018-05-18:
Upon using and installing the latest version of LuCI the web interface does not load, yielding this error message:
```
/usr/lib/lua/luci/dispatcher.lua:230: /etc/config/luci seems to be corrupt, unable to find section 'main'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:230: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>
```

This is irrespective of the actual luci configuration file used (/etc/config/luci), including the default auto-generated config.

jow-, 2018-05-19:
This can also mean that LuCI is unable to query uci data via ubus. It is likely caused by enabling uci session isolation.

Please provide the output of `ls -l /usr/share/rpcd/acl.d/` and `ls -l /usr/share/acl.d/`

closed 2018-05-19 by jow-

initramfs, 2018-05-19:
ls -la /etc/config/luci:
```
-rw-------    1 root     root          1078 May 18 22:33 /etc/config/luci
```
The config file itself (autogenerated):
```

config core 'main'
	option lang 'auto'
	option resourcebase '/luci-static/resources'
	option mediaurlbase '/luci-static/openwrt.org'

config extern 'flash_keep'
	option uci '/etc/config/'
	option dropbear '/etc/dropbear/'
	option openvpn '/etc/openvpn/'
	option passwd '/etc/passwd'
	option opkg '/etc/opkg.conf'
	option firewall '/etc/firewall.user'
	option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
	option sessionpath '/tmp/luci-sessions'
	option sessiontime '3600'

config internal 'ccache'
	option enable '1'

config internal 'themes'
	option Bootstrap '/luci-static/bootstrap'
	option OpenWrt '/luci-static/openwrt.org'

config internal 'apply'
	option rollback '30'
	option holdoff '4'
	option timeout '5'
	option display '1.5'

config internal 'diag'
	option dns 'lede-project.org'
	option ping 'lede-project.org'
	option route 'lede-project.org'

```

ls -l /usr/share/rpcd/acl.d/
```
-rw-r--r--    1 root     root           159 May  5 03:44 luci-base.json
-rw-r--r--    1 root     root           179 May  5 03:44 unauthenticated.json
```
ls -l /usr/share/acl.d/
```
-rw-r--r--    1 root     root            50 May  5 03:44 dnsmasq_acl.json
-rw-r--r--    1 root     root            91 May  5 03:44 luci-base.json
```

Note: I forgot to mention but this manifests as a regression for me, downgrading to a previous build with the same config files works just fine.

msangel, 2018-12-27:
I cant confirm

msangel, 2018-12-27:
root@OpenWrt:~# ls -la /etc/config/luci
-rw-r--r--    1 root     root           742 Dec 26 22:29 /etc/config/luci
root@OpenWrt:~# grep "main" /etc/config/luci
config core 'main'
root@OpenWrt:~# opkg list-installed luci*
luci - git-18.210.69179-6df9a57-1
luci-app-firewall - git-18.210.69179-6df9a57-1
luci-base - git-18.210.69179-6df9a57-1
luci-lib-ip - git-18.210.69179-6df9a57-1
luci-lib-jsonc - git-18.210.69179-6df9a57-1
luci-lib-nixio - git-18.210.69179-6df9a57-1
luci-mod-admin-full - git-18.210.69179-6df9a57-1
luci-proto-ipv6 - git-18.210.69179-6df9a57-1
luci-proto-ppp - git-18.210.69179-6df9a57-1
luci-theme-bootstrap - git-18.210.69179-6df9a57-1
root@OpenWrt:~# ls -l /usr/share/rpcd/acl.d
-rw-r--r--    1 root     root           159 Jul 30 16:25 luci-base.json
-rw-r--r--    1 root     root           179 Jul 30 16:25 unauthenticated.json
root@OpenWrt:~# ls -l /usr/share/acl.d/
-rw-r--r--    1 root     root            50 Jul 30 16:25 dnsmasq_acl.json
-rw-r--r--    1 root     root            91 Jul 30 16:25 luci-base.json
root@OpenWrt:~# cat /etc/config/luci

config core 'main'
	option lang 'auto'
	option mediaurlbase '/luci-static/bootstrap'
	option resourcebase '/luci-static/resources'

config extern 'flash_keep'
	option uci '/etc/config/'
	option dropbear '/etc/dropbear/'
	option openvpn '/etc/openvpn/'
	option passwd '/etc/passwd'
	option opkg '/etc/opkg.conf'
	option firewall '/etc/firewall.user'
	option uploads '/lib/uci/upload/'

config internal 'languages'

config internal 'sauth'
	option sessionpath '/tmp/luci-sessions'
	option sessiontime '3600'

config internal 'ccache'
	option enable '1'

config internal 'themes'
	option Bootstrap '/luci-static/bootstrap'

config internal 'diag'
	option dns 'lede-project.org'
	option ping 'lede-project.org'
	option route 'lede-project.org'

root@OpenWrt:~#

msangel, 2018-12-27:
note: the server is **lighttpd**
and configured as in this instruction:
http://ainopara.github.io/openwrt-webdav.html

hnyman, 2018-12-27:
```
note: the server is lighttpd
and configured as in this instruction:
http://ainopara.github.io/openwrt-webdav.html

> > 路由器系统：OpenWrt Barrier Breaker 14.07正式版

```

Note that you are following an ancient guide written for the year 2014 version of Openwrt & LuCI.

commit 9ed48ef2 (2018-05-19): luci-base: break circular luci.config <> luci.model.uci dependency

On certain environments, mainly with the embedded uhttpd interpreter, the
luci.config class cannot be loaded due to a circular dependency with the
luci.model.uci class.

Break up the dependency by deferring the loading of luci.config in
luci.model.uci until it is actually needed.

Fixes #1803, FS#1553.