openwrt/luci #1804: Unable to save WiFi password

49c29fa9, 2018-05-19:
Tplink 4300

When I enter this password +7|LeCNA'w(EW/QWLy\WCl6Pp:F3B8w5L*,wteTz~Bj~i`2&3G)5LUy6aLHr"$? as WiFi password, it doesn't save it. 

Basically, I can change password via telnet, thats about it but even then I cannot save that password. I have to use a weaker password. 

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '1'
        option path 'platform/ar934x_wmac'
        option htmode 'HT20'
        option country 'US'
        option legacy_rates '0'
        option hwmode '11g'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'COVFEFE'
        option key '1q2w3e4r5t'
        option encryption 'psk2+ccmp'
        option wpa_disable_eapol_key_retries '1'

Also, although this is a wireless N router, I cannot change it to 11n but it shows 11g.

jow-, 2018-05-19:
I cannot reproduce your problem. Please provide more details. When I copy-paste your password into LuCI I get the following results:

root@OpenWrt:~# cat /etc/config/wireless 

```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/ar933x_wmac'
	option htmode 'HT20'
	option disabled '0'
	option channel '1'
	option country 'US'
	option legacy_rates '1'

config wifi-iface
	option device 'radio0'
	option mode 'ap'
	option ssid 'xxx'
	option encryption 'psk2'
	option key '+7|LeCNA'\''w(EW/QWLy\WCl6Pp:F3B8w5L*,wteTz~Bj~i`2&3G)5LUy6aLHr"$?'
```

The generated hostapd config seems fine as well:

```
root@OpenWrt:~# cat /var/run/hostapd-phy0.conf 
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
beacon_int=100
channel=1

ieee80211n=1
ht_coex=0
ht_capab=[SHORT-GI-20][SHORT-GI-40][RX-STBC1][DSSS_CCK-40]

interface=wlan0
ctrl_interface=/var/run/hostapd
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=+7|LeCNA'w(EW/QWLy\WCl6Pp:F3B8w5L*,wteTz~Bj~i`2&3G)5LUy6aLHr"$?
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=xxx
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=14:e6:e4:e6:80:88
```

I am also able to copy-paste the same key into network-manager on Linux and get a working connection.

The hwmode 11g option does not relate to 11n capabilities, it specifies the frequency band to use.

49c29fa9, 2018-05-19:
Here are the settings:

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '1'
        option path 'platform/ar934x_wmac'
        option htmode 'HT20'
        option country 'US'
        option legacy_rates '0'
        option hwmode '11g'
        option require_mode 'N'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'COVFEFE'
        option key '6eJHrmBmxqKXcPLGYAakhHhP6WCFnEcuR1vzwr557gPX4ZmYkqm8m9e2zfzB
        option encryption 'psk2+ccmp'
        option wpa_disable_eapol_key_retries '1'

I can change the password if its hexadecimal characters but if its  ASCII characters I cannot. It doesn't work. 

I get the passwords from here https://www.grc.com/passwords.htm

jow-, 2018-05-19:
Please define `doesn't work`. I have no issues saving special passwords here. What exactly do you do that does not work?

49c29fa9, 2018-05-19:
When I save the password via Putty, the wireless gets disabled or turns into open network. 

When I save the password via Luci-ssl interface, it doesn't save at all.

49c29fa9, 2018-05-19:
OpenWrt SNAPSHOT, r6952-5399de7 // I didn't have this issue in the previous snapshot.

jow-, 2018-05-19:
How are you saving your password via putty? You will need to escape a number of cahracters when you set it via `uci set`. What means "doesn't save at all" ? Page refreshing with previous values restored?

49c29fa9, 2018-05-19:
I am saving 

vi /etc/config/wireless then I press i and edit it out. 

The page will show config saving and count down but after countdown the router restarts but the password is not saved but its still the old password.

jow-, 2018-05-19:
Did you verify that using Putty by running `cat /etc/config/wireless`? If you enter the password and just press change, is it shown correctly in the "unsaved changes menu"?

jow-, 2018-05-19:
Are you changing the password while connected via wireless?

49c29fa9, 2018-05-19:
I will do firstboot and change it again and get back to you.

49c29fa9, 2018-05-20:
Check these photos

https://img42.com/@Z7EIB

49c29fa9, 2018-05-20:
Changed the password via ethernet but its not saving at all.

ThomasCr, 2018-05-29:
are you sure you save your edited file correctly?

vi /etc/config/wireless
press i to enter inside mode
make your changes
press :x 
you should get be asked if you want to save your changes - press y

how looks your file after save your password with luci? what is the difference to your typed password?

ThomasCr, 2018-05-29:
ok, sorry again: it looks like a escape problem:

+7|LeCNA'\''w(EW/QWLy\WCl6Pp:F3B8w5L*,wteTz~Bj~i`2&3G)5LUy6aLHr"$?
is getting to
+7|LeCNA'w(EW/QWLy\WCl6Pp:F3B8w5L*,wteTz~Bj~i`2&3G)5LUy6aLHr"$?

try to use passwords without the ticks '

jow-, 2018-05-29:
The escaping is fine and performed automatically by uci.
E.g. `foo'bar` is getting stored as `'foo'\''bar'` and returned as `foo'bar` when read back.

jow-, 2018-07-03:
After looking at this again I see no indication of any actual problem, closing.

closed 2018-07-03 by jow-