openwrt/luci #1813: luci-app-ddns: EXTREME SLOWDOWN

Ansuel, 2018-05-23:
Ok i found all the reason of my issue recently with luci...

Luci-app-ddns was causing lot of lags to the entire interface. Every request (from the login) adds 10 second.
by removing this module and resetting cache in /tmp, loads time reduced to 100-200 ms

Some way to find what is slowing down it?

Ansuel, 2018-05-23:
ok i think i found the problem...

by removing this from the index function in controller/ddns.lua

the 10s delay is removed... 
think that luci create index for every module before the load of the actual interface (cache creation? )
and loads this every time... 
think this is the reason of the delay... a broken modle that slow down the entire system for every request, even the login one.
this looks wrong... why we need to loads the entry of every module if we are not in the actual page of it... 
it's just extra data to call... 
anyway i will check what is slowing down all in this file...
```
        local nxfs	= require "nixio.fs"		-- global definitions not available
	local sys	= require "luci.sys"		-- in function index()
	--local ddns	= require "luci.tools.ddns"	-- ddns multiused functions
	local muci	= require "luci.model.uci"
    --
	---- no config create an empty one
	--if not nxfs.access("/etc/config/ddns") then
	--	nxfs.writefile("/etc/config/ddns", "")
	--end
```

Ansuel, 2018-05-25:
Fixed with #1816 #1819

dawncold, 2018-05-27:
Thanks !@Ansuel 
I got same thing and I have no idea why luci is extreme slow down after I connect to a pptp server, and I'm debuging pptp or pppd...

I removed luci-app-ddns, all work fine even I connected to pptp server!!!

Ansuel, 2018-05-27:
i pushed a lot of optimizzation... if you need ddns module just make this modification locally 👍

Ansuel, 2018-06-05:
fixed.

closed 2018-06-05 by Ansuel

apnof, 2018-08-04:
Hi. 
I assume the fix is included in Release 18.06.0?
(As far as my understanding of the commits reaches, #1816 and #1819 were commited to master? (at the beginning of june))

Many thanks for the fix. 
Was running 17.01.4 the first time with an image from the imagebuilder and didn't know why luci was unusable (once i loaded the webpage, the router needed to be restarted...)

silence48, 2019-08-11:
thanks for this. i couldn't figure out why it was sooo SLOWWWW