openwrt/luci #1844: Problems moving firewall rules

kristrev, 2018-05-30:
While setting up some firewall rules and then re-ordering them (both traffic rules and port forwards), I noticed some issues when moving rules up/down. The issues caused the move(s) to trigger unexpected behavior.

* Sometimes, when clicking "Save & Apply" after I moved one or more rules, Luci throws the following error:

```
/usr/lib/lua/luci/dispatcher.lua:445: Failed to execute arcombine dispatcher target for entry '/admin/network/firewall/forwards'.
The called action terminated with an exception:
invalid key to 'next'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:445: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:127: in function </usr/lib/lua/luci/dispatcher.lua:126>
```

When checking the firewall configuration from the command line or returning to the Port Forwards/Traffic Rules page, I see that the order of the firewall rules has been updated.

* When I do not get the crash after clicking on "Save & Apply", then I see that no changes are made in the configuration. However, the "Unsaved changes" box appears and reports that there are unsaved changes. When I check these changes, I see that they match the updates I have done to the order of the rules.

* Sometimes, the changes reported in "Unsaved changes" after I click "Save & Apply" contains moves of rules that I have not touched. I double-checked that I for example had not moved one rule above another one.

If I click "Save" and then "Save & Apply" (from "Unsaved changes"), then the configuration is always updated correctly.

jow-, 2018-05-30:
Can you please check which version of `rpcd` is isntalled on the device?

kristrev, 2018-05-30:
Ah, sorry :) It is commit 820621952d537c49deba470c6f61c40df93f4ba8 (2018-05-13).

kristrev, 2018-05-30:
I have a separate fork of Luci with some local changes, and the most recent upstream commit is 658b6d7d2131b9dc891389b22ba35b4b937d9396.

However, I have seen the problem before this commit. I also tested with Luci master and I can see the error there. However, the errors does not always happen. For example, the exception is usually triggered on the first move after I have created a rule.

jow-, 2018-05-31:
Thanks, I tracked down the issue and working on a fix now. Luckily we can vastly improve the section reorder protocol now since the ubus uci api does not require section-by-section swaps anymore.

kristrev, 2018-05-31:
Thanks for the quick replies and great that you could track it down. If you need or want me to test something, just let me know.

jow-, 2018-05-31:
The following diff should solve the `next` crash, but I am still in the process of testing:

```diff
diff --git a/modules/luci-base/luasrc/cbi.lua b/modules/luci-base/luasrc/cbi.lua
index 472864211..4800d2aa7 100644
--- a/modules/luci-base/luasrc/cbi.lua
+++ b/modules/luci-base/luasrc/cbi.lua
@@ -1226,13 +1226,14 @@ function TypedSection.parse(self, novld)
                local stval = RESORT_PREFIX .. self.config .. "." .. self.sectiontype
                local order = self.map:formvalue(stval)
                if order and #order > 0 then
-                       local sid
-                       local num = 0
+                       local sids, sid = { }, nil
                        for sid in util.imatch(order) do
-                               self.map.uci:reorder(self.config, sid, num)
-                               num = num + 1
+                               sids[#sids+1] = sid
+                       end
+                       if #sids > 0 then
+                               self.map.uci:reorder(self.config, sids)
+                               self.changed = true
                        end
-                       self.changed = (num > 0)
                end
        end
 
```

kristrev, 2018-05-31:
Thanks! I will test it during the day, unless you beat me to it :) Will it also take care of the other issues, that moving rules can have an unpredictable outcome?

jow-, 2018-05-31:
I pushed the complete fix commit to my local fork.

kristrev, 2018-05-31:
I tested the commit now, and while the exception is gone, it seems that reordering rules has no effect?

I tested by creating four rules and manually reordering them, so that the order was as follows:
```1
3
4
2```

I then re-ordered them so that the order was as follows (in Luci):
```
2
3
4
1
```

After clicking "Save & Apply", the order was reverted to the original order. I also saw that "Unsaved changes" reported about three changes. However, these changes seems to be what was already done (restoring the previous order).

jow-, 2018-06-01:
I am unable to reproduce this on latest master, however this has significant changes since the commit you mentioned above.

Did you reload the page after applying the ordering? Since uci will rehash the anonymous sections on commit, the already rendered map will use deviating IDs, this is why I changed latest master to always reload the page after apply.

kristrev, 2018-06-01:
Turns out I was a bit tired last night when testing this, and I had applied your patch to my fork/branch. I am very sorry about that. When testing with the latest master, reordering rules works as expected. Thanks a lot for fixing the problem so fast!

jow-, 2018-06-01:
Thanks for confirming!

commit 2d352a92 (2018-06-03): luci-base: cbi: atomically reorder uci sections

Since the switch to ubus uci operations we do not have a local application-
side cursor cache anymore, instead uci operations happen synchronously in
the rpcd backend server.

This may cause cbi section reorder operations involving multiple elements
to fail, because anonymous section hashes may change due to rehashing
between consecutive ubus uci reorder calls.

In order to avoid that problem, use the ubus uci batch reorder extension,
which allows to pass a complete (or partial) list of section ids in the
desired order in one call, bypassing the [... cut]
closed 2018-06-03 by jow-