openwrt/luci #1849: Backup Restore broken - again

schlaile, 2018-06-02:
Trying to restore firmware configuration using

    /cgi-bin/luci/admin/system/flashops

will stall indefinitely and finally break with

    Bad Gateway
    The process did not produce any response

Adding debug code around the filehandler in system.lua around line 324 like this:

        dbg = io.open("/tmp/restore_debug.log", "w")    
                                                        
        http.setfilehandler(                                                    
                function(meta, chunk, eof)                                   
                        dbg:write("Hi there\n")                                
                        if not fp and meta and meta.name == "archive" then      
                                fp = io.open(archive_tmp, "w")                  
                        end                                                     
                        if fp and chunk then                                    
                                fp:write(chunk)                                 
                        end                                                     
                        if fp and eof then                                      
                                fp:close()                                 
                        end                                                     
                end                                                             
        )                                                                 
                                                                          
        dbg:close()                                                             

shows, that the filehandler is never called (restore_debug.log will be created, but no content is added to it).

Could you please take a look?

I'm a little bit lost in the http.lua callback handling mechanisms here...

schlaile, 2018-06-02:
I forgot to mention: 
I'm using lede-project.org current snapshot (which is lede master, right?)

schlaile, 2018-06-02:
okay, forget it.

I got the file permissions on the configuration archive wrong and Firefox is so incredible stupid, that it doesn't throw an error message and is instead just posting nothing or an empty file.

It kinda would be nice, if luci doesn't hang in that moment and says something in the lines of: well nice try, but I got an empty file?

jow-, 2018-06-04:
Hm, I am not able to reproduce this using some simple test upload with a zero-byte file. Is there any chance you could capture wireshark logs of the post request or maybe export the request as .har archive in the firefox debug console?

jow-, 2018-06-04:
Seems you've uncovered an interesting firefox bug here, basically Firefox sends a post request like this:

```POST /cgi-bin/luci/admin/system/flashops/restore HTTP/1.1
Host: 10.11.12.13
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:60.0) Gecko/20100101 Firefox/60.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Referer: http://10.11.12.13/cgi-bin/luci/admin/system/flashops
Content-Type: multipart/form-data; boundary=---------------------------9993942082038013279650384106
Content-Length: 499
Cookie: sysauth=5692d073a9da77c5df9bfcf4282ce92e
Connection: keep-alive
Upgrade-Insecure-Requests: 1

```

but without any HTTP message body, which is clearly wrong. It's `Content-Length` header even indicates that there's 499 bytes of data to follow...

Will add a workaround.

jow-, 2018-06-04:
Turns out that we cannot fix it in LuCI, the problem is in uhttpd. Since Firefox neither sends a request body, nor shuts down the connection, the LuCI process will time out while attempting to read amount of data advertised in `Content-Length` ... needs some fix on the uhttpd side most likely.

schlaile, 2018-06-04:
I wanted to file a bug in the Firefox bugtracker and just noticed, that well, I'm round about 10 years late... :(

https://bugzilla.mozilla.org/show_bug.cgi?id=426177

Maybe someone will notice now, that there is actually a problem with Firefox
and: @jow- thanks for trying to work around the problem in LuCI / uhttpd...

I agree, that you can't really fix the problem here (okay: maybe we could change the timeout error to something in the lines of: "the request timed out and there should be a file, but there isn't, bahaha, you are using Firefox, stupid! :)" )

jow-, 2018-06-25:
I've tested this situation with other setups, such as using a dummy CGI script under Apache and it will result in an ungraceful Gateway Timeout server error there as well, so closing this as wontfix.

closed 2018-06-25 by jow-