openwrt/luci #1852: pptp

m0rDev, 2018-06-04:
I'm sorry if I write wrong.

LEDE Reboot 17.01.4 r3560-79f57e422d

Model | TP-Link TL-WDR4300 v1
Firmware Version | LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)

kmod-pptp | 4.4.92-1
ppp-mod-pptp | 2.4.7-12

If you set the parameter ** ipparam ** in file /etc/ppp/options.pptp
then in Web lucy -> network -> Interfaces - the status pptp of the connection is not displayed

RX: 0 B (0 Pkts.)
TX: 0 B (0 Pkts.)

This connection exists. It can be seen through the terminal - ifconfig

pptp-usd Link encap:Point-to-Point Protocol
inet addr:192.168.100.52 P-t-P:192.168.100.240 Mask:255.255.255.255
UP POINTOPOINT RUNNING NOARP MULTICAST MTU:1500 Metric:1
RX packets:3073 errors:0 dropped:0 overruns:0 frame:0
TX packets:5025 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:3
RX bytes:314513 (307.1 KiB) TX bytes:525333 (513.0 KiB)

If you remove or comment out the parameter ** ipparam ** in file /etc/ppp/options.pptp. Then in the Web Lucy -> network -> Interfaces pptp status is working

Uptime: 0h 0m 13s
MAC-Address: 00:00:00:00:00:00
RX: 64 B (4 Pkts.)
TX: 96 B (5 Pkts.)
IPv4: 192.168.100.58/32

jow-, 2018-06-04:
Yes, do'n't set `ipparam` manually, it is set automatically by netifd to track the interface state.

closed 2018-06-04 by jow-

m0rDev, 2018-06-04:
You did not answer anything

m0rDev, 2018-06-04:
If you do not specify an option with an **ipparam**, then the script is not executed

yousong, 2018-06-05:
@m0rDev , the network stats and state info in LuCI mostly comes from netifd.  netifd protocol handler will pass `ipparam <netname>` argument to the pppd command line so that the ifup/ifdown script can tell netifd the event source.  That means you are not supposed to pass arbitrary `ipparam` of your own when working with netifd.

please see in the source code for details `package/network/services/ppp/files/ppp.sh`, `package/network/services/ppp/files/lib/netifd/ppp-up`

m0rDev, 2018-06-05:
@yousong I did not touch the source code. I specify in the settings of ppp /etc/ppp/options.pptp

**/etc/ppp/options.pptp**

noipdefault
noauth
nobsdcomp
nodeflate
idle 0
#mppe required,no40,no56,stateless
nomppe
maxfail 0
usepeerdns
ipparam kelokepptpd
mtu 1436
mru 1436

**/etc/ppp/options**

debug
logfile /dev/null
noipdefault
noaccomp
nopcomp
nocrtscts
lock
maxfail 0
lcp-echo-failure 5
lcp-echo-interval 1

**/etc/ppp/ip-up.d/script**

```
#!/bin/sh
# parameters
# $1 the interface name used by pppd (e.g. ppp3)
# $2 the tty device name
# $3 the tty device speed
# $4 the local IP address for the interface
# $5 the remote IP address
# $6 the parameter specified by the 'ipparam' option to pppd
case "$6" in
 kelokepptpd)
  /sbin/route add -net 192.168.100.0 netmask 255.255.255.0 dev $1
  /usr/sbin/iptables -I OUTPUT -d 192.168.100.0/24 -o $1 -j ACCEPT
  /usr/sbin/iptables -I INPUT -s 192.168.100.0/24 -i $1 -j ACCEPT 
  /usr/sbin/iptables -I FORWARD -d 192.168.100.0/24 -o $1 -j ACCEPT 
  /usr/sbin/iptables -I FORWARD -s 192.168.100.0/24 -i $1 -j ACCEPT 
  /usr/sbin/iptables -t nat -A POSTROUTING -o $1 -j MASQUERADE

  ;;
 *)
esac
exit 0
```

**/etc/ppp/ip-down.d/script**

```
#!/bin/sh
# parameters
# $1 the interface name used by pppd (e.g. ppp3)
# $2 the tty device name
# $3 the tty device speed
# $4 the local IP address for the interface
# $5 the remote IP address
# $6 the parameter specified by the 'ipparam' option to pppd

case "$6" in
 kelokepptpd)
  /sbin/route del -net 192.168.100.0 netmask 255.255.255.0 dev $1
  /usr/sbin/iptables -D OUTPUT -d 192.168.100.0/24 -o $1 -j ACCEPT
  /usr/sbin/iptables -D INPUT -s 192.168.100.0/24 -i $1 -j ACCEPT
  /usr/sbin/iptables -D FORWARD -d 192.168.100.0/24 -o $1 -j ACCEPT
  /usr/sbin/iptables -D FORWARD -s 192.168.100.0/24 -i $1 -j ACCEPT
  /usr/sbin/iptables -t nat -D POSTROUTING -o $1 -j MASQUERADE

  ;;
 *)
esac   
exit 0 
```

jow-, 2018-06-05:
Yes, as I already wrote - you must not specify a custom ipparam. OpenWrt will supply its own to identify ppp sessions. Overriding it through /etc/ppp/* will break that.

jow-, 2018-06-05:
> He does not define anything. There's clearly something wrong.

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/services/ppp/files/ppp.sh#l133