openwrt/luci #1859: Log file not being written to for second ddns updater

cryobry, 2018-06-06:
The first ddns configuration entry log is working fine. However, the second ddns configuration entry log is not being written to. The log is being created at /var/log/ddns/host_ipv4.log, but it is just a series of blank lines.

hnyman, 2018-06-06:
Sounds like a bug in the ddns itself, not LuCI, so this LuCI bug tracker is likely wrong place.
cc @chris5560 @Ansuel

Ansuel, 2018-06-06:
@cryobry umm i have 2 configuration , one for ipv6 and one for ipv4 and i can correctly see log for both of them... 
describe your configuration pls

cryobry, 2018-06-06:
I have two ipv4 ddns entries:

`cat /etc/config/ddns

config ddns 'global'
	option ddns_dateformat '%F %R'
	option ddns_loglines '250'
	option upd_privateip '0'

config service 'router_ipv4'
	option service_name 'namecheap.com'
	option lookup_host 'router.snipped.com'
	option domain 'snipped.com'
	option username 'router'
	option password 'snipped'
	option ip_source 'web'
	option interface 'wwan'
	option enabled '1'
	option check_interval '60'

config service 'htpc_ipv4'
	option service_name 'namecheap.com'
	option lookup_host 'htpc.snipped.com'
	option domain 'snipped.com'
	option username 'htpc'
	option password 'http://checkip.dyndns.com'
	option ip_source 'web'
	option interface 'wwan'
	option enabled '1'
	option check_interval '60'
`

/var/log/ddns/router_ipv4.log contains the typical logged output

/var/log/ddns/htpc_ipv4.log contains a growing number of blank lines

cryobry, 2018-06-06:
Rechecked my configuration and there was an error:

```
cat /etc/config/ddns

config ddns 'global'
	option ddns_dateformat '%F %R'
	option ddns_loglines '250'
	option upd_privateip '0'

config service 'router_ipv4'
	option service_name 'namecheap.com'
	option lookup_host 'router.snipped.com'
	option domain 'snipped.com'
	option username 'router'
	option password 'snipped'
	option ip_source 'web'
	option interface 'wwan'
	option enabled '1'
	option check_interval '60'

config service 'htpc_ipv4'
	option service_name 'namecheap.com'
	option lookup_host 'htpc.snipped.com'
	option domain 'snipped.com'
	option username 'htpc'
	option password 'http://checkip.dyndns.com'
	option ip_source 'web'
	option interface 'wwan'
	option enabled '1'
	option check_interval '60'
```
In the second config I must've erroneously pasted 'http://checkip.dyndns.com' instead of my password token. I changed it to my password and now I have correct logging. I'm posting this here in case you want to catch this problem somehow.

Cheers.

closed 2018-06-06 by cryobry