openwrt/luci #1860: luci 802.11ac wifi channels listed incorrect for width size selected

sirjaymz, 2018-06-07:
luci does not update to the correct available channel list when selecting the width size of the operating frequency for a Qualcomm Atheros QCA9880 802.11nac (Archer C7), and maybe other chipsets as well.

These lists are mutually exclusive, meaning, channels identified on 20Mhz width list are not duplicated in other width lists. Same goes with 40Mhz width's are not duplicated, as well as 80Mhz, and the 160Mhz width lists.

**When selecting a 20 Mhz width size, the following channels should be listed.**
All allowable channels should be listed per regulatory. This list is currently what luci displays for ALL (20, 40, 80) frequency widths, which is incorrect. This is correct for the 20Mhz width size only. [Ch (Mhz)]
36 (5180), 40 (5200), 44 (5220), 48 (5240), 52 (5260), 56 (5280), 60 (5300), 64 (5320), 100 (5500), 104 (5520), 108 (5540), 112 (5560). 116 (5580), 120 (5600), 124 (5620), 128 (5640), 132 (5660), 136 (5680), 140 (5700), 144 (5720), 149 (5745), 153 (5765), 157 (5785), 161 (5805), 165 (5825)

**When selecting a 40 Mhz width size, only these channels should be displayed.**
38 (5190), 46 (5230), 54 (5270), 62 (5310), 102 (5510), 110 (5550), 118 (5590), 126 (5630), 134 (5670), 142 (5710), 151 (5755), 159 (5795)

**When selecting a 80 Mhz width size, only these channels should be displayed.**
42 (5210), 58 (5290), 106 (5530), 122 (5610), 138 (5690), 155 (5775)

**When selecting a 160 Mhz width size, only these channels should be displayed.**
Not sure if this needs to be included, as the Archer C7 does not support 80+80 or 160 frequency widths but, other chipsets may support it, so maybe beneficial to fix it in luci anyway.
50 (5250)
114 (5570)

The list should be updated, dependent upon which country the user has selected as the Regulatory body and laws require.

As an example,  New Zealand and the USA follow literally an identical frequency range, [here is an example](https://web.archive.org/web/20170802164314/https://www.rsm.govt.nz/consumers/pdf-and-documents-library/Wi-Fi%20Devices%20using%20the%205%20GHz%20Band.pdf) of how the frequencies and the overlaps are rolled up and visually identifiable.

Reference : [List of WLAN channels](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax))

sirjaymz, 2018-06-07:
jow - Thanks for the reply.
Maybe I could look at how the current list is created. Can you point me to which files I should be looking at to see how the current list is generated?

jow-, 2018-06-07:
Sure, take a look here: https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/view/cbi/wireless_modefreq.htm

sirjaymz, 2018-06-08:
Upon further investigation, I was able to determine the driver allows for dynamic assignments of channels, but also, this will allow for programmatically setting each of the lists, according to the details provided by each PHY. 

`iw phy0 channels`

displays the channel number, the width that channel supports and the maximum tx power supported by the phy. 

`iw reg get`

displays the country based permissible frequencies, along with the speed (htmode) of which those frequencies support. Additionally this yields which frequency ranges are actually allowed to utilize, as well as which frequencies support Auto Bandwidth assignments.

For the programmatically side of the things, we will have to determine the best way to parse the data into the most universal method. I only but have the single 5.0ghz router/ap device type, archer C7 to test with, as I am curious on how other devices with different driver, such as a broadcom, actually assigns channels and frequencies. I will expand my router inventory to take a look. (new project = new tools :) )

Upon reading and getting a better understanding of how the 'ath10k' driver works, the 80 mhz and the 160 mhz window sizes are dynamically flexible, based on the driver. This allows for the window to slide to different combination of the frequencies to make a the 80 mhz window, instead of it hard coding, or limiting, the window to the actual 802.11x 5.0ghz defined channels.

Some low hanging fruit I think I see , is that luci doesn't actually set the 
    CONFIG_IEEE80211AC
    CONFIG_ACS
commands in the hostap config to enable the radar and dynamic frequency selections for this driver at least. 

I am messing with this to see if I can make some progress towards implementing a solution with some additionally settings that could be added to allow for some low level higher performing settings without allot of effort.

hopefully i understand this well enough to actually make an impact. JS is still relatively new to me, but seems like a straight forward language.

feckert, 2019-10-09:
We haven't heard back from you, so we're going to go ahead and close the issue.
If you're still experiencing the problem, please re-open the issue. There were a lot of changes recently in master. So please try this or the upcoming release branch 19.07

closed 2019-10-09 by feckert