openwrt/luci #1897: While Flashing the firmware, its taking  so much time to flash if keep settings is disabled.

vsaichandra, 2018-06-21:
Hi,

I am trying to flash a firmware from luci -> system -> Backup/Flash firmware.

- if KEEP SETTINGS is disabled while flashing, even after the flashing is completed its still showing the "flashing page"

- Is there any solution for this case, so that when the flashing is completed the page gets reloaded properly. 

Thank you.

jow-, 2018-06-21:
There is no solution to this. I assume your LAN ip changes when you do not keep settings so the browser is never able to reconnect to the device. This is not really solvable unless you access your OpenWrt via hostname exclusively.

vsaichandra, 2018-06-21:
Thank you for the information :)

hnyman, 2018-06-21:
> if KEEP SETTINGS is disabled while flashing, even after the flashing is completed its still showing the "flashing page"

If you are using HTTPS version of LuCI:

That may also be the browser reacting to the changing SSL certificate (as a new certificate is automatically re-generated at the first boot after flash if settings were not kept). 

That happens for me with Firefox. I need to refresh the page after the flash and then the new HTTPS session is started.  (If I keep settings, then also the certificate is kept and all works ok)

jow-, 2018-07-03:
Wontfix.

closed 2018-07-03 by jow-