openwrt/luci #1899: ipv6 interface address reported incorrectly in luci

justdave, 2018-06-21:
The "IPv6 suffix" setting (ip6ifaceid in the /etc/config/network file) on an interface is correctly honored when configuring the interface, but is not correctly shown in the luci interface statistics.
In the GUI, the interface's address is always show as if ip6ifaceid was set to ::1, even if the interface isn't actually using that address.

[image]

[image]

```
root@WiFi5G:~# ifconfig eth0.54
eth0.54   Link encap:Ethernet  HWaddr 10:6F:3F:E7:79:62
          inet addr:192.168.54.3  Bcast:192.168.54.255  Mask:255.255.255.0
          inet6 addr: fe80::126f:3fff:fee7:7962/64 Scope:Link
          inet6 addr: fd7a:fe7f:6a03:54::fe/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:1036 (1.0 KiB)
```

```
root@WiFi5G:~# opkg list_installed | grep luci
luci - git-18.147.69097-36945b5-1
luci-app-firewall - git-18.147.69097-36945b5-1
luci-base - git-18.147.69097-36945b5-1
luci-lib-ip - git-18.147.69097-36945b5-1
luci-lib-jsonc - git-18.147.69097-36945b5-1
luci-lib-nixio - git-18.147.69097-36945b5-1
luci-mod-admin-full - git-18.147.69097-36945b5-1
luci-proto-ipv6 - git-18.147.69097-36945b5-1
luci-proto-ppp - git-18.147.69097-36945b5-1
luci-ssl - git-18.147.69097-36945b5-1
luci-theme-bootstrap - git-18.147.69097-36945b5-1
```

jow-, 2018-06-22:
Dupe of #1484 and fixed in master with 461df8b0dc1a9be5fb9e904d8cdc44c6d0cbd6c8

closed 2018-06-22 by jow-

justdave, 2018-06-22:
Awesome, thanks!