openwrt/luci #1903: luci-app-ddns utilizes Old Wiki Links

lleachii, 2018-06-22:
Reported by : OxR463 on forum.lede-project.org

https://forum.lede-project.org/t/applications-luci-app-ddns-opennic-integration/15867

`
applications/luci-app-ddns.`

> Also, that package has several references to the old wiki.
> It would be nice to get those updated to the new site.

Old Wiki links appear:
```

Dynamic DNS allows that your router can be reached with a fixed hostname while having a dynamically changing IP address.
OpenWrt Wiki: DDNS Client Documentation --- DDNS Client Configuration
```
Links to:

http://wiki.openwrt.org/doc/howto/ddns.client
and
http://wiki.openwrt.org/doc/uci/ddns

Change to:

https://openwrt.org/docs/guide-user/services/ddns/client (FYI, there is also a "to merge" page that exists on the new WIki)
and
https://openwrt.org/docs/guide-user/base-system/ddns

Ansuel, 2018-07-16:
Fixed the referenced pr

oxr463, 2018-11-19:
Looks like we can close this, (Yes, I created a GitHub account).

closed 2018-11-19 by hnyman