openwrt/luci #1904: Hitting Submit on Startup page does not save /etc/rc.local

lleachii, 2018-06-23:
https://forum.lede-project.org/t/startup-not-saved-in-luci-18-06-branch-18-06-rc1/15920

When I edit and save the /etc/rc.local file via LuCI (i.e. on the System > Startup page)...hitting Submit has no effect.

Current workaround: edit file on the command prompt

This bug is on 18.06.0-rc1.

This does not persist in master, where I can tell the difference by observing the newly designed Overview page.

lleachii, 2018-06-23:
https://github.com/openwrt/luci/issues/1900

jow-, 2018-07-02:
How large is the content you're trying to save?

westislander, 2018-07-06:
I am experiencing this same problem with a 269 line file of 8068 bytes I attempt to paste in the "startup" box. It seems that I can paste somewhere around 40 lines but more than that won't get saved. The same file works just fine on openwrt 17.01.4 .

I am using a TP-LINK TL-WDR3600 with Lede version
 OpenWrt 18.06.0-rc1 r7090-d2aa3a1b62 / 			LuCI openwrt-18.06 branch (git-18.163.60998-c379f00)

lleachii, 2018-07-06:
@jow- 

> root@OpenWrt:~# ls -l /etc/rc.local 
> -rw-r--r--    1 root     root          6636 Jun 29 14:25 /etc/rc.local

westislander, 2018-07-07:
Saving to "Scheduled tasks" also resulted in nothing being saved when I tried my 8068 byte text file. While I realize this is rather large for a "crontab" file, I wanted to see if there was a common problem, and indeed there was.

westislander, 2018-07-07:
I did some Further testing, an 852 byte "scheduled tasks" file can be saved successfully.
When I increase the file size to much over 4007 bytes it starts to fail.
For me, the problem is size dependent. It is preventing me from using my startup scripts.
All this was tested using a TP-Link TL-WDR3600
The image file name is:
openwrt-18.06.0-rc1-ar71xx-generic-tl-wdr3600-v1-squashfs-sysupgrade.bin

Here is an excerpt from my kernel log identifying the hardware:
[    0.000000] Linux version 4.9.109 (buildbot@builds-03.infra.lede-project.org) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r6984-fa0275b) ) #0 Thu Jun 21 09:22:05 2018
[    0.000000] MyLoader: sysp=84c31530, boardp=0065cbe4, parts=195ef6a4
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9344 rev 2

dibdot, 2018-07-07:
This is a known issue and fixed in master, probably we need to backport 80cb4fe

jow-, 2018-09-18:
Fixed by https://github.com/openwrt/luci/commit/80cb4fef8c7db0dadc373fef122d7abb092a7191

closed 2018-09-18 by jow-