openwrt/luci #1918: Interface device for ping -I <> not properly resolved for alias interfaces.

pleasantone, 2018-06-29:
If you try to diagnose a v6 interface in mwan3 such as pinging the tracking or gateway IP, we get back `ping: bad address '@wan`.

This isn't an edge-case, this is the current default documented setup for dual-stack interfaces. If you have a dual-stack interface using the v4/v6 alias convention as per the default documentation:

```
config interface 'wan'
	option ifname 'eth0.10'
	option proto 'dhcp'
	option metric '120'

config interface 'wan6'
	option ifname '@wan'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option metric '120'
	option reqprefix '60'
```

You get:
```
root@g2s:/usr/sbin# uci get network.wan6.ifname
@wan_cc
root@g2s:/usr/sbin# uci get network.wan.ifname
eth0.10
```

The problem is here:

https://github.com/openwrt/luci/blob/485e5c930ff0a98f1155b04080f6736270c58a10/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L141

but I don't know the proper fix to issue a pull request. Seems like either UCI should be fully resolving option ifnames that have @ in them, or we should be doing a second resolution step to check to see that we have a real interface instead of an alias.

@feckert

jow-, 2018-06-29:
```
local nm = require "luci.model.network".init()
local network = nm:get_network(interface)
local device = network and network:ifname()

if device then
    ...
end
```

pleasantone, 2018-06-30:
Tested the patch from @jow- it is great except we cannot delete the 'local uci =' line because we reference 'uci' later in the function. I haven't set up a clone of packages for pull requests yet to make the pull request, but I'll try to do so in the next 24hrs if someone else hasn't already submitted it.

Thank you very much to @jow- for the awesome explanation of "the right way" to do it.

pleasantone, 2018-07-02:
Pull request #1929

jow-, 2018-07-03:
Fixed.

closed 2018-07-03 by jow-