openwrt/luci #1919: luci-app-lxc: add option "-B best" to "lxc-create"

localnet, 2018-06-29:
@dibdot 

Adding the option "-B best" to "lxc-create" it try to find the best backingstore.

From [documentation](https://linuxcontainers.org/fr/lxc/manpages/man1/lxc-create.1.html):

> **-B, --bdev** _backingstore_
> 'backingstore' is one of 'dir', 'lvm', 'loop', 'btrfs', 'zfs', 'rbd', or 'best'. 
> [...]
> If backingstore is 'best', then lxc will try, in order, btrfs, zfs, lvm, and finally a directory backing store.

In my case I am creating the LXC containers in a disk with filesystem BTRFS and a subvolume "@lxc" mounted in "/srv/lxc" with the command:
`lxc-create -n server -t download -B best -- -d alpine -r edge -a armhf --no-validate`

The result is the next:
```
# btrfs subvolume list /srv/lxc
ID 257 gen 151666 top level 5 path @lxc
ID 260 gen 151666 top level 257 path server/rootfs
```

dibdot, 2018-06-30:
@localnet Thanks for your suggestion. Please test attached (untested) PR.

localnet, 2018-06-30:
I am having troubles with "lxc_get_arch_target()", I getting error message:

> There are no templates for your architecture (armv7l) available, please select another containers URL.

I expect "armv7l" be translated to "armhf" to I can use containers from "images.linuxcontainers.org".

Thanks you.

dibdot, 2018-06-30:
> I expect "armv7l" be translated to "armhf" to I can use containers from "images.linuxcontainers.org".

This breaks the turris repo which uses "armv7l" ... therefore I've changed the code and use the mapping for linuxcontainers.org only.

dibdot, 2018-06-30:
Thanks for testing. The backingstore support via ubus seems not to work - remove it for now.
Regarding turris repo:

* This repo is only available via https (therefore index-system.2)
* This repo only supports target armv7l (and not arm7hf)

Nevertheless, I'll push my PR to fix the mapping for linuximages.org.

localnet, 2018-06-30:
> This repo is only available via https (therefore index-system.2)

I will try to review if I have all required to https (I am using ca-bundle it is a dependency of msmtp and not ca-certificates)

> This repo only supports target armv7l (and not arm7hf)

I tried with target armv7l, I edited and corrected my previous comment when I did it

Please test the view (lxc.htm) before push, I see it strange I don’t sure if is what you want (I am using openwrt git tag v18.06.0-rc1 as source to build)

Thanks you.

dibdot, 2018-07-01:
> Please test the view (lxc.htm) before push, I see it strange I don’t sure if is what you want (I am using openwrt git tag v18.06.0-rc1 as source to build)

That’s expected … you shouldn't test current master frontend changes with 18.06. Few weeks ago, there was a treewide LuCI change to convert HTML tables to div based markups (together with additional styling changes). These changes are not in openwrt 18.06 tree and of course not in turris tree.

localnet, 2018-07-01:
Excuse me, my fault. I have few experience trying to collaborate.

Thanks for your work.

PS: Feel free to close the issue, or maintain it open to investigate the backingstore support via ubus. As you prefer.

dibdot, 2018-07-01:
I will not give up so easily ...;-)
I've reworked the lxc-create call - should work now. Would be nice if you could test this second attempt. Thanks!

localnet, 2018-07-01:
Now it is working:
  - I create a new container from LUCI (test)
```
# btrfs subvolume list /srv/lxc
ID 257 gen 151755 top level 5 path @lxc
ID 258 gen 151755 top level 257 path test/rootfs
```
Thanks you!

dibdot, 2018-07-01:
Feature added with 1f2d40b. Thanks!

closed 2018-07-01 by dibdot