openwrt/luci #1943: Adding a hostname does not restart dnsmasq

austinbutler, 2018-07-07:
I found https://github.com/openwrt/luci/issues/453, but it was closed saying it should be resolved.

This still happens for me on `LuCI lede-17.01 branch (git-17.315.61305-a9bddc4)`, I have to go in and restart manually with `/etc/init.d/dnsmasq restart`.

dibdot, 2018-07-07:
Works for me. Please re-test with current snapshots or at least with 18.06 branch. In any case this is not a LuCI issue.

closed 2018-07-07 by dibdot

austinbutler, 2018-07-07:
@dibdot would you mind letting me know which project would be responsible if I test and find the same issue with 18.06?

dibdot, 2018-07-08:
dnsmasq is an openwrt core package and the issue tracker is here: https://bugs.openwrt.org

Quickly checked the gitlog of and found the relevant commit ;-)

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4f02285d8b4a66359a8fa46f22a3efde391b5419

So, this will be fixed in 18.06.