openwrt/luci #1948: NGINX File dependency error

fuqiang03, 2018-07-14:
[image]
[image]

build000, 2018-07-21:
@hnyman

Ansuel, 2018-07-21:
are you sure you have updated the package feeds?

build000, 2018-07-21:
Situation to openwrt-18.06...no master.
Yes - updated feeds

Ansuel, 2018-07-21:
openwrt-18.06 doesn't have the new changes to nginx... think that's the problem...
instead luci have the new package....

(so i think luci package should be dropped or nginx changes should be pushed upstream)

build000, 2018-07-21:
... well ... there is no change for 18.06...

jow-, 2018-07-21:
https://github.com/openwrt/luci/commit/14c73faf313a85f9a70b21b1afd84a47e28a93c0

closed 2018-07-21 by jow-

rossiniscarface, 2019-01-25:
Looks like in the 18.06-snapshots there are still luci-nginx and luck-nginx-ssl packages built.
https://downloads.openwrt.org/releases/18.06-SNAPSHOT/packages/arm_cortex-a5_neon-vfpv4/luci/luci-nginx_git-18.201.34715-c813598-1_all.ipk

hnyman, 2019-01-26:
> Looks like in the 18.06-snapshots there are still luci-nginx and luck-nginx-ssl packages built.

They are not built any more, but the old binaries have not been deleted.
@jow-