openwrt/luci #1959: 18.06.0-rc2: Luci-ssl not starting upon opkg install

tobyknudsen, 2018-07-17:
Howdy,
Upon upgrade to RC2, I then ran 'opkg install luci-ssl' but it did not perform as usual:
- I recall only HTTP server was up and authentication was broken.
- Luci-ssl worked properly upon reboot.
Sorry to not have more details. Ask me if you have any questions.
T

jow-, 2018-07-17:
As far as I recall, installing `luci-ssl` never restarted the webserver, it simply pulls in a cert generator and a crypto library. Maybe you used to install `luci-ssl` from scratch, without a prior running gui?

tobyknudsen, 2018-07-17:
Hi Jo-Philipp, I could be mistaken but I recall different behavior ,with
RC2 being less convenient.
'opkg install luci-ssl' would previously install the service, I had
generated cert previously, and it would start the service.
With RC2 it would not restart the service, only http, but a reboot brought
it up right.
I encountered this issue once. Hope that helps.

jow-, 2018-07-17:
That was never encoded behaviour of the package, but as I said, maybe you previously installed it on a system where no uhttpd was installed yet? In this case it would automatically start it and build a cert.

tobyknudsen, 2018-07-17:
Ok. Perhaps I have not actually reported anything, but I have updated the
router (keeping configuations) around 20 times and this was the only time
that required a reboot after the opkg install.

Keep up the good work!

closed 2018-10-20 by jow-