openwrt/luci #1977: adblock: can't save empty whitelist

skath, 2018-07-21:
adblock - 3.4.3-1
luci-app-adblock - git-18.180.55366-b78664c-1

In /luci/admin/services/adblock/advanced/whitelist, add some domains to the whitelist and save. 

Then, try deleting the entries and replacing it with an empty whitelist. Every time you click 'save' the previous whitelist replaces the desired empty whitelist. 

As a workaround I had to add a comment line to the file to remove all whitelisted domains since an empty one can't be saved.

dibdot, 2018-07-21:
Delete your whitelist entries, hit "Enter" and click save ...  and you're done.

closed 2018-07-21 by dibdot

jow-, 2018-07-21:
@dibdot - Might be a case of `.write()` vs. `.remove()` in cbi, see e.g. 0251603f0aa064109138d40d06aa892f49e7abc7