openwrt/luci #1988: LUCI isn't deleting some entries when 'DHCP Client' and 'DHCPv6 Client' mode is enabled

ShapeShifter499, 2018-07-23:
Supply the following if possible:
 - Device problem occurs on
WD My Net n750
TP-Link TL-WDR3600 v1

 - Software versions of OpenWrt/LEDE release, packages, etc.

LEDE Reboot 17.01.4 r3560-79f57e422d 

 - Steps to reproduce
   
From fresh install go to Network > Interfaces and edit the LAN interface.  Switch the protocol to 'DHCP CLient' and click switch protocol under 'Really switch protocol?'  After it switches then click 'Save & Apply' near the bottom.   

Checking the /etc/config/dhcp  file after this shows some leftover settings that should have been removed but were not.

Attached are two different /etc/config/dhcp files, one labelled dhcp-incorrect.txt which is what results after preforming the above steps and another labelled dhcp-correct.txt which I believe should be the correct settings that should be applied after preforming the above steps.

[dhcp-correct.txt](https://github.com/openwrt/luci/files/2219533/dhcp-correct.txt)
[dhcp-incorrect.txt](https://github.com/openwrt/luci/files/2219534/dhcp-incorrect.txt)

ShapeShifter499, 2018-07-27:
Has anyone taken the time to look at this issue?

jow-, 2018-07-28:
Yeah I looked at it and will fix it after the release. Since the `dnsmasq` init script will refuse to operate on non-static interface (https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/services/dnsmasq/files/dnsmasq.init;h=c1ae0934fd2c466687c80f7ae7f6fa567e4474fa;hb=7a6b2badfa56cfd833d64f113398198f71fd896e#l477) and since one can disable the DHCP server before switching the protocol I do not deem it release critical yet.

ShapeShifter499, 2019-04-19:
@jow- it's been awhile and this issue hasn't been closed. Has this been fixed for newer releases of OpenWRT and LUCI?

jow-, 2019-04-20:
No, it hasn’t been fixed.

adelton, 2025-01-10:
For the record, the diff of those two files (sans whitespace) is
```
--- /tmp/dhcp-correct.txt	2025-01-10 01:48:28.594506176 +0100
+++ /tmp/dhcp-incorrect.txt	2025-01-10 01:48:30.030509808 +0100
@@ -17,7 +17,12 @@
 
 config dhcp 'lan'
         option interface 'lan'
-        option ignore '1'
+        option start '100'      
+        option limit '150'      
+        option leasetime '12h'  
+        option dhcpv6 'server'      
+        option ra 'server'          
+        option ra_management '1'    
 
 config dhcp 'wan'
         option interface 'wan'
```

@ShapeShifter499 You say that you switched the interface to DHCP Client. I assume the original protocol for the interface was Static address -- is that correct?

Do those DHCP server related options functionally cause a problem, when they are left in the `/etc/config/dhcp`? Or is this mostly a cosmetic thing?

I wonder if LuCI does not actually try to be helpful here. When the user switches a protocol and then realize that they made a mistake and switch back, they will find most of the original settings back.