openwrt/luci #2044: 18.06.0 Bad Gateway on System->Startup page

JamesT42, 2018-08-07:
Hi!
I get a Bad Gateway error message when I want to access System->Startup. All other pages are fine. Restart does not help. No clue in system log.
Any ideas?
Cheers
James

jow-, 2018-08-08:
Please check the system log for uhttpd errors.

JamesT42, 2018-08-08:
There an none, except the login accepted messages. I have logread -f running while looking for the problem, nothing pops up during the error.

JamesT42, 2018-09-12:
Still there with 18.06.1
Can I enable some kind of debugging output?

JamesT42, 2018-12-04:
The only thing we seem to have in common is openvpn.

napoxog, 2018-12-09:
> Did you try to remove openvpn package and check the startup page? I'll check later today and let you know the result.

On my side removal and re-installation of openvpn package did not solve the problem

RudeDude-420, 2019-01-02:
I encountered this issue myself after upgrading my WNDR3400v1 to 18.06.1, and after some trial and error I have some unfortunate news. This issue seems to stem from the proprietary broadcom driver kmod-brcm-wl. After performing a clean flash and installing a few packages at a time and rebooting, each time checking for the issue, something in this sequence of changes caused the issue. After removing the broadcom driver it is now working. I really hope I'm wrong because it's pretty much the best driver for my platform.
```
boot completed
installed usb-modeswitch
installed luci-app-statistics
installed nas
installed kmod-brcm-wl
installed wl
installed wlc
installed luci-app-nlbwmon
installed adb
installed ipset
removed kmod-b43
removed kmod-mac80211
removed kmod-cfg80211
installed bwm-ng
installed iptables-mod-tee
(accidentally) upgraded luci-mod-admin-full
reboot
```
if anyone wants the full console output/history let me know.

JamesT42, 2019-01-02:
I'm certain that I did not have the kmod-brcm-wl installed. Maybe it is a random error?

JamesT42, 2019-01-02:
> 
> 
> @JamesT42 which version of luci-mod-admin-full are you using? -- from 18.06.1 or from master?

18.06.0 and 18.06.1

Digitint, 2019-04-01:
I may confirm that same is happening immediately after installing kmod-brcmsmac driver on 18.06.2. I also saw "kern.err kernel: brcmsmac bcma0:1: START: tid 1 is not agg'able" error in system log.

nukkkkk, 2019-04-20:
H

ph03n1xSVJ, 2019-07-28:
I have been bumping into this for 3 days now, and am narrowing this down to either
1 - Setting by MAC-Filter for both the 2.4 GHz and 5GHz channels
2 - Installing shadowsocks-libev, specifically ss-config, ss-rules, ss-redir and luci-app-shadowsocks-libev. I am NOT installing ss-redir, ss-server and ss-tunnel.

My router is WNDR3700v4 and am running 18.06.4. There is nothing in the log that indicates something is wrong with uhttpd. I will try narrowing down between these two one of these days...

jow-, 2020-02-23:
Not able to reproduce on 19.07 / master

closed 2020-02-23 by jow-