openwrt/luci #2046: "connect" button in interface enables bring up on boot

widewing, 2018-08-08:
Expected behavior: just bring up the interface once, should not alter config and make the interface auto connect. 

The button is confusing. People can always set "bring up on boot" in advance tab. Someone like me just want to bring interface up once and don't want it keep retrying, and don't want it connect on boot.

jow-, 2018-08-08:
Thanks, will change it to enable/disable when I find the time.

jow-, 2018-08-08:
Reconnect will be made available despite `auto=0` and Connect / Stop will be renamed to Enable/Disable. Due to the lack of a proper `enabled` uci option in netifd this is the only way to realize this functionality by the means of uci changes in conjunction with apply/rollback.

widewing, 2018-08-08:
Ok.. maybe just add an third button, that become "reconnect" | "connect"/"disconnect" | "enable"/"disable"

feckert, 2019-08-11:
This is not an issue its a feature request to add an additional button reconnect. So please open a feature request issue for this feature

closed 2019-08-11 by feckert