openwrt/luci #2050: Wireguard interface shown as IPv4 upstream instead of WAN interface

jorti, 2018-08-08:
OpenWrt 18.06.0 x86_64
luci-ssl - git-18.212.55608-5ec7261-1

After configuring a Wireguard interface, Luci shows the Wireguard interface as IPv4 upstream, instead of the real WAN interface.

https://ibb.co/jVZqQp

jow-, 2018-08-08:
LuCI shows the first interface carrying a default route as upstream interface.

closed 2018-08-08 by jow-

jorti, 2018-08-08:
But my default route is via eth1

```
root@OpenWrt:~# ip r
default via 90.162.x.x dev eth1 proto static src 90.162.x.y
90.162.56.0/24 dev eth1 proto kernel scope link src 90.162.x.y
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev eth2 proto kernel scope link src 192.168.2.1 
192.168.3.0/24 dev eth3 proto kernel scope link src 192.168.3.1 
192.168.4.0/24 dev vpn proto kernel scope link src 192.168.4.1 
```

jorti, 2018-08-09:
You are right, unchecking "Route Allowed IPs" in the WireGuard peer, and rebooting fixes the problem.

Thank you.