openwrt/luci #2052: Previous Luci version number is cached after update

mbo2o, 2018-08-10:
After updating Luci Packages the Luci Version displayed is not the installed version but the previous version.

If the cache files are removed manually then the correct version is displayed
.
`rm /tmp/luci-modulecache/* /tmp/luci-indexcache`

Issue was observed on 17.01.x and 18.06.1

Reference:
https://forum.openwrt.org/t/solved-luci-revision-numbers/17590

hnyman, 2018-08-10:
Hmmm.
The files in /tmp will not survive flash & reboot, so you can only run into this if you update LuCI and do not reboot.

Maybe "rm those files" should be added into luci-base postinst function.

jow-, 2020-02-23:
Not able to reproduce on any version I tested

closed 2020-02-23 by jow-