openwrt/luci #2056: 'Open ports on router’ option dissapears when there is not firewall zone called 'wan'

shoulders, 2018-08-11:
Hi

FIrst of all I have been configuring my BT Home Hub v5A router to be a wireless client that gives a NAT network on the 4 yellow ethernets and my own private network through the red socket. This all works fine but the reason I tell you this is that I renamed all of the zones so they were relevant to my setup hence why I renamed the default **wan** to **nat_wwan**

**Open ports on router** missing  (Network --> Firewall --> Traffic Rules)

when I rename the **wan** zone to **nat_wwan** the option 'Open ports on router’ option dissapears. I can get it to reappear by either renaming this zone back or creating a new zone with the name **wan**.

Also rules I have added do work even if this option is not here. (i.e. I added them before renaming the zone)

How it should look
[image]

What it looks like without the **wan** zone
[image]

So basically there is some sort of dependance on having a zone named **wan**, can this be fixed/removed.

My Setup
- Firmare version  = OpenWrt 18.06.0 r7188-b0b5c64c22 / LuCI openwrt-18.06 branch (git-18.210.69179-6df9a57) - (18.06 Stable)
- Router = BT Home Hub 5A (it is actually a Plusnet one)
- if my configuration is of note I can supply that aswell

Thanks

dibdot, 2018-08-12:
This should work most probably with a named zone section, e.g. (untested!):

```
config zone 'wan'
        option name 'nat_wwan'
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        option masq '1'
        option mtu_fix '1'
        option network 'trm_wwan'
```

shoulders, 2018-08-12:
@dibdot 

This is a workaround that you have to use the SSH to fix rather than using than using LuCI.

thanks anyway

shoulders, 2018-08-13:
@jow- thanks for the quick update

for reference I am running my OpenWRT as a wireless client (x2) which I named **nat_wwan** and **bridge_wwan** which are both wan connections so to keep things neat I used appropriate names and deleted the **wan** interface.

commit f64b1523 (2018-08-16): luci-app-firewall: allow "open ports" when no wan zone exists

Arguably this makes little if no wan zone exists but prefer consistency
over heuristics and always render the "open port" shortcut.

Fixes #2056

(cherry picked from commit bf123fb7c4918120f5f9f638866bfae1e5b23a54)