openwrt/luci #2065: OUI lookup in the DHCP leases table?

rlipscombe, 2018-08-14:
I've got a lot of devices that appear as '?' in the DHCP leases table. It'd be really useful if LuCI could look up the OUI and tell me the manufacturer. Ideally this'd be integrated in the table directly as an extra column, but I could live with an extra click.

jow-, 2018-08-23:
OUI prefix data is *huge* so the lookup would require online access. Neither is ideal.

49616e, 2019-06-25:
At the time of writing this comment, the OUI table available at [here](http://standards-oui.ieee.org/oui/oui.txt) about 4072KB in size. We can trim it down to less than half the size since we don't really need those addresses anyway. There are plenty of devices right now that comes with 32MB or bigger flash, I think <2MB is not that big of a deal if made available as an optional package. Please reconsider adding support for this as it'll be very nice to have in certain situations.

CTrabant, 2020-01-19:
As of this comment, the combined OUI lookup table from wireshark:
https://code.wireshark.org/review/gitweb?p=wireshark.git;a=blob_plain;f=manuf

is *1,632 KB*, and is an assemblage of the following sources:
```
# The IEEE public OUI listings available from:
# <http://standards-oui.ieee.org/oui/oui.csv>
# <http://standards-oui.ieee.org/cid/cid.csv>
# <http://standards-oui.ieee.org/iab/iab.csv>
# <http://standards-oui.ieee.org/oui28/mam.csv>
# <http://standards-oui.ieee.org/oui36/oui36.csv>
```

That size shouldn't be much of a problem, especially if a lookup were an opt-in feature to turn on.

ervee, 2023-03-12:
If I would download the OUI file on my AP or Router (both are pretty modern and have hundreds of MB storage available), how would I make use of this? I would love to see the MAC lookup used in the "Wireless Associated Stations" and "Active DHCP Leases"  view?

joshenders, 2023-03-12:
> That size shouldn't be much of a problem, especially if a lookup were an opt-in feature to turn on.

I think we could probably do even better with a binary format, as the OUI bytes could be stored as binary data and the strings compressed. Opt-in is icing on the cake.

n-thumann, 2023-12-10:
In case this get tackled, I'd appreciate it getting added to the list of associated stations (connected wireless devices) too :v:

systemcrash, 2023-12-10:
The most realistic way of handling this involves _compression_. Whatever format the list is in, a lookup would not suffer to run a zstdgrep, zcat or whatever binary to dig for a matching prefix and stash those in an intermediate cache.

I don't know how much caching should be done to prevent continuous lookup for the same devices, but some amount seems necessary.

For those who want stuff like this, what makes it so attractive?

kzaoaai, 2023-12-10:
For me it helps identify devices without hostnames, like IOT devices

CTrabant, 2023-12-11:
> For those who want stuff like this, what makes it so attractive?

For me to more easily identify anything from IoT devices, wireless speakers, guest laptops, etc.

n-thumann, 2023-12-11:
> For those who want stuff like this, what makes it so attractive?

I'm running APs with OpenWrt and having the vendor name next to connected wireless device eases identification.

vulpes2, 2025-05-08:
Wrote a simple script to condense the three OUI csv files required for manufacturer identification into minified json, the file size is around 1.4MB. It can be served by uhttpd, and lookup should be performed with client side js. The json file can be compressed with gzip (456KB) or brotli (400KB), but unless uhttpd implements custom header support (https://github.com/openwrt/uhttpd/pull/13) and/or MIME detection, compressed files cannot be served with the correct `Content-Type` header and thus cannot be decompressed by the browser.

```
https://standards-oui.ieee.org/oui/oui.csv
https://standards-oui.ieee.org/oui28/mam.csv
https://standards-oui.ieee.org/oui36/oui36.csv
```

Now the question is where the OUI information should be displayed. Obvious places are the overview page (DHCP leases) and the wireless page (associated stations). However, both of the lists are already very crowded, adding another column that can potentially contain long text probably wouldn't look too good. The alternative solution is to build a dedicated luci app, and it could pull MAC addresses from the v4/v6 route tables, active dhcp leases and associated stations (if wireless support is present) and combine them all into a single list, probably with another column to indicate where the MAC address is pulled from as well.

I'd like to implement this at some point, @jow- can you provide some input on where you'd prefer this to be placed in the UI? Should I implement this in luci-base or just make a separate luci app instead?

ervee, 2025-05-09:
My 2ct: A dedicated luci app would be nice, but IMHO it should be possible on the main page.
If I look at my main luci page I see in "Active DHCP Leases" the "Hostname" switching between ```devicename``` and ```devicename (hostname.domain.tld)```.
And under "Associated Stations" the "Host" column switches between ```hostname (IP address)``` and ```hostname.domain.tld (IP address)```.

So I would not mind it the "MAC address" column would switch between ```MAC address``` and ```OUI``` with the latter possibly truncated to fit. So it would display ```00:1c:7f:69:69:69``` a while and then ```Check Point So...```. And that could be enhanced with the full name in a mouse over ```?``` like the port status.

commit 70b7176f (2025-09-25): luci-mod-status,luci-mod-network: support oui to vendor resolving

For easier definition of connected devices, this commit adds support
to identify them by vendor name.

Package `upf-neigh` is needed to lookup the vendor name in a hash table.
It implements a ubus-call `ubus call fingerprint fingerprint`:

```
root@<redacted> ~ # ubus call fingerprint fingerprint
{
        "7c:c2:55:XX:XX:XX": {
                "vendor": "Super Micro Computer, Inc."
        }
}
```

Fixes: #2065
Depends on: openwrt/packages#27257
closed 2025-09-25 by systemcrash