openwrt/luci #2073: After adding client wireless interface, i am not able to edit master interface.

joewander, 2018-08-17:
Hi

In Network -> WiFi tab, i am not able to edit the master interface after adding client interface.

steps:
1. Add a client interface in 2.4ghz.
2. Click on edit for the master interface in 2.4ghz.
3. Its throwing an error as shown below.

**Error**
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute arcombine dispatcher target for entry '/admin/network/wireless/wifi0.network1'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/admin_network/wifi.lua:163: attempt to call method 'is_up' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167> 

Can you please help me on understanding whats causing this error and how to solve it ?

Thank you..!

jow-, 2018-08-17:
Please provide the exact version of LuCI installed.

jow-, 2018-08-17:
This release is several years old, I cannot provide support for it anymore.

closed 2018-08-17 by jow-