openwrt/luci #2078: Luci unable to resolv IPv6 hostnames

th0ma7, 2018-08-18:
On my wifi access point Luci status page is able to resolv and show the IPv4 hostnames such as:
mydesktop.lan (192.168.80.46)

Although two issues:
1) it is unable to resolv IPv6 only hosts and will only show it's IPv6 address instead of the hostname
2) for systems that has both IPv4 & IPv6 addresses it would be great to have both addresses showned such as: mydesktop.lan (192.168.x.y, fe80::5660:9ff:fef4:d638)

Note that my routeur resolv.conf file is pointing to /tmp/resolv.conf.auto so it's able to resolv using my main routeur DNS resolver.  And using host command (from bind-host package) i'm fully able to resolv all IPv4 & v6 addresses & names on my LAN.

Thnx in advance.

jow-, 2018-08-18:
Can you resolv the hostname with „nslookup ipv6addr“ ?

th0ma7, 2018-08-18:
Indeed I can:
```
# nslookup fd44:adb8:e3f4::56b
Server:		192.168.80.1
Address:	192.168.80.1#53

b.6.5.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.4.f.3.e.8.b.d.a.4.4.d.f.ip6.arpa	name = eclipse.lan
```

Although this is rather interesting as the said "hostnames" which I first thought where IPv6 may be something else entirely?

[image]

th0ma7, 2018-08-19:
Further investigating, while DHCP on my wifi access-point was disabled it was still offering dhcp6 addresses thus equipments where getting fd7e:* from my wifi AP instead of the fd44:* from my main router.

Now that this is solve, still the status pages shows IPv6 link-local fe80:: adresses instead of the fd44:* address it should have received.  But I'm starting to wonder if that is related to Android devices behaving weird with IPv6 compared to my linux box which works alright?

jow-, 2018-08-20:
fe80:* addresses are shown for devices not performing DHCPv4 or DHCPv6 - such devices will auto-assign an IP themselves using stateless autoconfig. Due to the nature of SLAAC, these auto-assigned IPs cannot be really seen on the router, only the link local IPv6 addresses can be inferred from `ip -6 neigh` and this is what LuCI is displaying, since the fe80:* IP is the only known piece of information for the given wireless client MAC.

So in short, this is nothing that can be fixed since the information simply isn't available anywhere on the router.

closed 2018-08-20 by jow-