openwrt/luci #2098: User login fail on controller with non root users after upgraded to 18.06

shawnchain, 2018-08-19:
Hi, 
I have created a controller index pages and  specified  a system user (not root) for access authentication.
```
local page = entry({"mmdvm", "service"}, form("mmdvm/service_form"), _("Service") , 2)
page.sysauth = "mmdvm"
page.sysauth_authenticator = "htmlauth"
```
which works perfectly under 17.01 with existing user 'mmdvm' .

But this trick does not work under 18.06. 
When I access the protected pages under 18.06,  I was asked for the password for above user, with 'mmdvm" filled in the user form field, everything seems fine but when I type the password and press login, "invalid username or password" message returns.

logread shows:
```
daemon.err uhttpd[1610]: luci: failed login on /mmdvm/service for mmdvm from 192.168.19.xxx
```

Any hints?

Thanks,

Shawn

shawnchain, 2018-08-19:
cat /etc/os-release shows

```
root@OpenWrt:/usr/lib/lua/luci/controller/mmdvm# cat /etc/os-release
NAME="OpenWrt"
VERSION="18.06.1"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06.1"
VERSION_ID="18.06.1"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r7258-5eb055306f"
LEDE_BOARD="brcm2708/bcm2708"
LEDE_ARCH="arm_arm1176jzf-s_vfp"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06.1 r7258-5eb055306f"
```

jow-, 2018-08-20:
Make sure you have an `mmdvm` account declared in `/etc/config/rpcd`

shawnchain, 2018-08-20:
Hi jow,

Problem seems solved by adding following section into `/etc/config/rpcd`
```
config login
        option username 'mmdvm'
        option password '$p$mmdvm'
        list read '*'
        list write '*'
```
But I'm still confused with 2 questions:
 1. Will this setup introduce security issue to  my openwrt box ?
 2. Why 18.06 needs this additional setup compare to 17.01

Thanks,

Shawn

jow-, 2018-08-20:
1. No, the above section basically defines another user account allowed to login via `ubus call session login`, no other system service uses these credentials - so SSH etc. are unaffected
2. This is due to the fact that LuCI on 17.01 had its own user authentication code directly checking `/etc/shadow` while LuCI on 18.06 switched to in-memory ubus based sessions provided by `rpcd`.

shawnchain, 2018-08-20:
I will close this issue as it's a by-design behavior. 
Thank you jow for your quick reply!

closed 2018-08-20 by shawnchain