openwrt/luci #2101: CBI form now has implicit dependency to the luci-mod-admin-full package.

shawnchain, 2018-08-20:
I have a CBI form which will change some UCI settings by a non-root user. It works fine under 17.01 with following entry setup:
```
page = entry({"mmdvm", "settings","general"}, cbi("mmdvm/general_settings",{hideapplybtn=true,hideresetbtn=true}), _("General Settings") , 30)
```
While under 18.01, the above setup will not commit the changes as `autoapply` is not set on the entry parameter. So I changed the above line to 
```
page = entry({"mmdvm", "settings","general"}, cbi("mmdvm/general_settings",{hideapplybtn=true,hideresetbtn=true,autoapply=true}), _("General Settings") , 30)
```
Now the problem occurs: 
 1. The auto apply only works for root user. 
 2. It requires  `controller/admin/uci.lua` which is provided by `luci-mod-admin-full`

Any hints ?

Thanks,

Shawn

shawnchain, 2018-08-23:
In short for this issue, If I choose the default CBI templates in my own Luci module for load/save/edit UCI config, I may have to keep the luci-mod-admin-full package installed or there will be an error.

jow-, 2018-08-23:
We're working on making `luci-base` sufficient to properly host applications without needing to have `luci-mod-admin-full` installed. In the meanwhile you do require this implicit dependency.

No immediate idea about the autoapply as root user, this is something I almost never test. Will eventually need to look into it.

shawnchain, 2018-08-23:
Thanks jow for your outstanding work!
Based on my test, autoapply works just fine as non-root user on my module. What I did is just 
```
m = Map("mmdvm", translate("General Settings"),
    translate("You can setup MMDVM general parameters here")
)
m.apply_on_parse = true -- this magic parameter takes the rescue!
```

jow-, 2018-09-20:
I believe your original issue is now addressed with https://github.com/openwrt/luci/commit/6ec0353201435e0d0d7d32820d8ba600b4ca7b5b, please reopen if there's still issues regarding the implicit dependency.

closed 2018-09-20 by jow-