openwrt/luci #2106: LUCI goes away w/18.06.0 & 18.06.1 MT7621A/AR71xx

stevehss, 2018-08-21:
Xiaomi Mi 3G (Mt7621A) :

After submitting a simple LAN IP address change Luci would not be accessible . SSH remains working.

TP-Link TL-WR1043v5 is having the same issue on 18.06.0 , did not confirm it on 18.06.1

jow-, 2018-08-21:
Please provide the output of `netstat -nltp` and `ps ww | grep uhttpd` in the broken state.

cshoredaniel, 2018-08-21:
Silly question, but are you browsing to the new address instead of the old address?  If you change the LAN IP then the (by definition) the address of the router changes.

stevehss, 2018-08-21:
Yes, I tried to access the new address. SSH is listening on the new port.

netstat result:

root@OpenWrt:~# netstat -ntlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      1631/dnsmasq
tcp        0      0 192.168.11.1:53         0.0.0.0:*               LISTEN      1631/dnsmasq
tcp        0      0 192.168.19.199:53       0.0.0.0:*               LISTEN      1631/dnsmasq
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      1111/dropbear
tcp        0      0 ::1:53                  :::*                    LISTEN      1631/dnsmasq
tcp        0      0 fe80::7a11:dcff:fe4c:b96b:53 :::*                    LISTEN      1631/dnsmasq
tcp        0      0 fe80::7a11:dcff:fe4c:b96c:53 :::*                    LISTEN      1631/dnsmasq
tcp        0      0 fe80::7a11:dcff:fe4c:b96b:53 :::*                    LISTEN      1631/dnsmasq
tcp        0      0 :::22                   :::*                    LISTEN      1111/dropbear

ps ww | grep uhttpd result is:

1706 root                  1204 S              grep uttpd

192.168.11.1 is refusing to connect to be exact but still, luci cannot be accessed.

jow-, 2018-08-21:
Ok, looks like the `uhttpd` process crashed then. Can you provide the output of `dmesg` and `logread` as well?

stevehss, 2018-08-21:
dmesg:

root@OpenWrt:~# dmesg
[    0.000000] Linux version 4.14.63 (buildbot@builds-03.infra.lede-project.org) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7102-3f3a2c9)) #0 SMP Thu Aug 16 07:51:15 2018
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3G
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 80545e80, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from 0x80549740 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu @81210000 s26256 r8192 d22896 u57344
[    0.000000] pcpu-alloc: s26256 r8192 d22896 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00035804
[    0.000000] Readback ErrCtl register=00035804
[    0.000000] Memory: 253664K/262144K available (4596K kernel code, 239K rwdata, 564K rodata, 220K init, 270K bss, 8480K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007814] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073976] pid_max: default: 32768 minimum: 301
[    0.078744] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085252] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094046] Hierarchical SRCU implementation.
[    0.099181] smp: Bringing up secondary CPUs ...
[    0.112677] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.112687] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.112699] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.112837] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.163930] Synchronize counters for CPU 1: done.
[    0.206796] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.206805] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.206815] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.206884] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255052] Synchronize counters for CPU 2: done.
[    0.293814] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.293823] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.293831] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.293911] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340234] Synchronize counters for CPU 3: done.
[    0.370099] smp: Brought up 1 node, 4 CPUs
[    0.377858] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.387643] futex hash table entries: 1024 (order: 3, 32768 bytes)
[    0.393939] pinctrl core: initialized pinctrl subsystem
[... 198 lines cut ...]
[   13.673010] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   13.680237] bus=0x2, slot = 0x1, irq=0xff
[   13.684329] PCI: Enabling device 0000:00:01.0 (0004 -> 0006)
[   13.690190] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   13.721932] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   13.833352] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   13.838842] mt76x2e 0000:02:00.0: Build: 1
[   13.842949] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   13.871233] mt76x2e 0000:02:00.0: Firmware running!
[   13.877633] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   13.885339] PPP generic driver version 2.4.2
[   13.891060] NET: Registered protocol family 24
[   13.898133] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.503317] urandom_read: 5 callbacks suppressed
[   14.503331] random: jshn: uninitialized urandom read (4 bytes read)
[... cut ...]
[   20.261356] device eth0 entered promiscuous mode
[   20.268118] br-lan: port 1(eth0.1) entered blocking state
[   20.273582] br-lan: port 1(eth0.1) entered disabled state
[   20.279506] device eth0.1 entered promiscuous mode
[   20.287542] br-lan: port 1(eth0.1) entered blocking state
[   20.293013] br-lan: port 1(eth0.1) entered forwarding state
[   20.299005] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   21.147161] random: crng init done
[   21.281427] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

stevehss, 2018-08-21:
logread:

root@OpenWrt:~# logread
Tue Aug 21 14:07:52 2018 kern.notice kernel: [    0.000000] Linux version 4.14.63 (buildbot@builds-03.infra.lede-project.org) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7102-3f3a2c9)) #0 SMP Thu Aug 16 07:51:15 2018
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 3G
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Determined physical RAM map:
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000]  memory: 10000000 @ 00000000 (usable)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Aug 21 14:07:52 2018 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Aug 21 14:07:52 2018 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Aug 21 14:07:52 2018 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Zone ranges:
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000]   HighMem  empty
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Early memory node ranges
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80545e80, node_mem_map 81003000
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000]   Normal zone: 512 pages used for memmap
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Tue Aug 21 14:07:52 2018 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x80549740 with crng_init=0
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu @81210000 s26256 r8192 d22896 u57344
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000] pcpu-alloc: s26256 r8192 d22896 u57344 alloc=14*4096
Tue Aug 21 14:07:52 2018 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
Tue Aug 21 14:07:52 2018 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Writing ErrCtl register=00035804
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Readback ErrCtl register=00035804
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Memory: 253664K/262144K available (4596K kernel code, 239K rwdata, 564K rodata, 220K init, 270K bss, 8480K reserved, 0K cma-reserved, 0K highmem)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.007814] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.073976] pid_max: default: 32768 minimum: 301
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.078744] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.085252] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.094046] Hierarchical SRCU implementation.
Tue Aug 21 14:07:52 2018 kern.info kernel: [    0.099181] smp: Bringing up secondary CPUs ...
Tue Aug 21 14:07:52 2018 kern.warn kernel: [    0.112677] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... cut ...]
Tue Aug 21 15:30:23 2018 authpriv.notice dropbear[1688]: Password auth succeeded for 'root' from 192.168.11.162:50025
Tue Aug 21 15:31:10 2018 daemon.warn dnsmasq-dhcp[1631]: no address range available for DHCP request via eth0.2
Tue Aug 21 15:33:50 2018 daemon.warn dnsmasq-dhcp[1631]: no address range available for DHCP request via eth0.2
Tue Aug 21 15:33:53 2018 daemon.warn dnsmasq-dhcp[1631]: no address range available for DHCP request via eth0.2
Tue Aug 21 15:37:01 2018 daemon.warn dnsmasq-dhcp[1631]: no address range available for DHCP request via eth0.2

cshoredaniel, 2018-08-21:
Did you modify /etc/config/uhttpd  at all? (e.g. to put the specific old LAN address(es)?

stevehss, 2018-08-21:
No. I just modded the LAN IP address from the GUI.

The router is accessible on the new IP once restarted but after modifying the IP address nothing comes back from the GUI.

From SSH I did modify it using the previous 18.06.0 version and it worked. However with the new 18.06.1 and did not try SSH since we develop firmwares to use coova-chilli captive portal. However the GUI should also work. Even if I know the workaround this is a flaw in stability and the end user won'zt be able to use SSH.

jow-, 2018-08-21:
What happens if you issue `/etc/init.d/uhttpd start` via SSH can you reach LuCI again?

stevehss, 2018-08-21:
Nope, same result "192.168.11.1 refused to connect."

cshoredaniel, 2018-08-21:
Have you changed the firewall?  Or perhaps it's something to do with Coova (since it's a captive portal)?

stevehss, 2018-08-22:
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^

stevehss, 2018-08-22:
It'a almost empty and this is not a configuration.. so it's either corrupted or gone for good

jow-, 2018-08-22:
This is an indication of a filesystem corruption. Did you at some point power cycle your router without rebooting?

stevehss, 2018-08-22:
Yes at some point since I was willing to get back the router. What time frame are we willing to see for the router to change a LAN IP address? How many seconds or minutes?
OpenWRT 12 and 14 and 15 and 17 were able to do this in a few seconds and give back access on GUI right away. We have been developing custom firmwares since then. So even if I did this to the file system, is this normal?

jow-, 2018-08-22:
Note that current LuCI will roll back the changes if connection to the router is lost. You do need to wait for the rollback dialog to appear and hit "apply unchecked" or - in recent master versions - simply reconnect your browser to the new IP address within the rollback grace period (usually 30s). Example video: http://luci.subsignal.org/~jow/rollback.mp4

As for the filesystem corruption on yanking power; this is an OpenWrt / Kernel / ubifs issue unrelated to LuCI.

jow-, 2018-08-23:
Closing this due to being unrelated to LuCI. The NAND / UBIFS filesystem corruption needs to be solved in OpenWrt base.

closed 2018-08-23 by jow-

riaqn, 2019-01-08:
Is there anyway to disable this "feature"?