openwrt/luci #2112: LCP echo failure threshold 0  does not behave as described in LuCI

bill888uk, 2018-08-25:
For BT Home Hub 5A configured to use PPPoE or PPPoA protocol for example.

In **LuCI→Network→Interfaces→WAN→Advanced Settings,** there are two default settings displayed:

LCP echo failure threshold    0
LCP echo interval    5

On the page, the 'default' setting **LCP echo failure threshold** to Zero, implies all LCP echo failures would be ignored. (see attached image) But I found this to be incorrect/misleading. 

There is no **keepalive** option found in the WAN interface section of /etc/config/network on Home Hub 5A when selecting PPPoE or PPPOA protocols.  But there appears to be a fixed timeout (possibly 5 seconds) on the Home Hub 5A when I use PPPoE on VDSL connection.  ie. not infinite as quoted in LuCI.

I don't know if this is the same behaviour for all devices.

If **LCP echo failure threshold** is set to a non-Zero value, then **keepalive** option is CREATED in the WAN interface within /e/c/network file.

The description displayed in LuCI for LEDE and OpenWRT 18: 'Presume peer to be dead after a given amount of LCP echo failures, **use 0 to ignore failures**' perhaps needs to be changed?

eg. perhaps simplest solution is to remove the phrase **'use 0 to ignore failures'**  ?

This closed ticket may be of interest  (I added my observations affecting HH5A running LEDE 17 to end of the closed ticket)
[https://bugs.openwrt.org/index.php?do=details&task_id=1259](url)

Would it be possible to add  'keepalive_adaptive' setting into LuCI mentioned in above ticket in the future?

[image]

jow-, 2018-08-30:
From `ppp.sh`:

```
        local lcp_failure="${keepalive%%[, ]*}"
        local lcp_interval="${keepalive##*[, ]}"
        ...
        [ "${lcp_failure:-0}" -lt 1 ] && lcp_failure=""
        ...
        proto_run_command "$config" /usr/sbin/pppd \
                ${lcp_failure:+lcp-echo-interval $lcp_interval lcp-echo-failure $lcp_failure $lcp_adaptive} \
```

Means if you set it to `0` in LuCI, `option keepalive` will be deleted. When no `option keepalive` is present, `lcp_failure will end up being empty which means that no `lcp-echo-interval` and no `lcp-echo-failure` is passed to pppd.

jow-, 2018-08-30:
Default values in pppd `lcp.c`:

```
int     lcp_echo_interval = 0;  /* Interval between LCP echo-requests */
int     lcp_echo_fails = 0;     /* Tolerance to unanswered echo-requests */
```

So I am confused. The option seems to behave as intended?

jow-, 2018-08-30:
To be perfectly clear: `0` is *supposed* to disable LCP stuff. If that does not work it is a bug that has to be fixed in OpenWrt.

jow-, 2018-08-30:
Turns out the OpenWrt implementation is completely flawed and cannot be fixed in a way where omitting `option keepalive` would yield the expected behaviour without breaking the majority of existing setups.

Will change LuCI to emit an explicit `option keepalive 0` but that'll still need fixes on the OpenWrt side.

kiral6, 2021-12-02:
from master `ppp.sh`:

        [ -n "$keepalive" ] || keepalive="5 1"
        
        local lcp_failure="${keepalive%%[, ]*}"
        local lcp_interval="${keepalive##*[, ]}"
        local lcp_adaptive="lcp-echo-adaptive"

In the LuCI setting, the default value is to leave blank, so $keepalive is null. At this time, lcp-failure=5, lcp-interval=1, this is inconsistent with the description of LuCI. And if you fill in 0/5 in LuCI, you can't save it, because the value hasn't changed and doesn't need to be saved, then $keepalive will always be null, which means that you can't set lcp-echo-failure = 0 through LuCI.

commit 1b2065d0 (2018-08-30): luci-proto-ppp: align keepalive defaults with current OpenWrt

Disabling LCP echos by *not* setting "option keepalive" was broken in
OpenWrt since https://dev.archive.openwrt.org/ticket/2373.html so setting
"0" in LuCI had the effect of reverting back to "5, 1" while the help
suggested otherwise.

Support for "keepalive 0" was fixed with https://git.openwrt.org/555c59230
so align LuCI now to emit "option keepalive 0" instead of removing the
option when entering "0" in the gui.

Fixes #2112
commit a388be0f (2019-07-19): luci-proto-ppp: align keepalive defaults with current OpenWrt

Disabling LCP echos by *not* setting "option keepalive" was broken in
OpenWrt since https://dev.archive.openwrt.org/ticket/2373.html so setting
"0" in LuCI had the effect of reverting back to "5, 1" while the help
suggested otherwise.

Support for "keepalive 0" was fixed with https://git.openwrt.org/555c59230
so align LuCI now to emit "option keepalive 0" instead of removing the
option when entering "0" in the gui.

Fixes #2112

(cherry picked from commit 1b2065d0be7f74cb93fac7f208ee4f28666d3f69)