openwrt/luci #2130: 500 internal server error while flashing the image

joewander, 2018-09-03:
HI,

While flashing the image, i am getting 500 internal server error as shown below.

ERROR:

<!DOCTYPE html>
<html lang="auto">
<head>
<meta charset="utf-8">
<title>Status: 500 Internal Server Error
/usr/lib/lua/luci/dispatcher.lua:433: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/material/header'.
A runtime error occured: [string "/usr/lib/lua/luci/view/themes/material/head..."]:79: attempt to index local 'boardinfo' (a nil value)
stack traceback:
[C]: in function 'assert'
/usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch'
/usr/lib/lua/luci/dispatcher.lua:168: in function </usr/lib/lua/luci/dispatcher.lua:167>

-- Can you help me understand why do we get such error and how to avoid it ?

dibdot, 2018-09-03:
Which LuCI version did you use?

dibdot, 2018-09-04:
CC is outdated and no longer supported. As a workaround you can always use sysupgrade directly in a ssh session. Please don't open more CC tickets - thanks!

closed 2018-09-04 by dibdot

hnyman, 2018-09-04:
15.05.x is pretty much unsupported.

> Failed to execute template 'themes/material/header'. 

Also, "material" theme does not exist in 15.05.
You should likely first reset theme to a supported theme (bootstrap and openwrt)

maricarcrisol, 2019-11-26:
What does 'CC' mean?